RetroPie/RetroPie-Setup #544: Permission error over a fresh install

mzeyrek, 2015-01-01:
Hi,

I've downloaded Noobs 1.3.11 which comes with Raspbian 2014-12-24. With a fresh install, once i set autostart emulationstation, gives below permission errors. I've already selected permission fix during setup and re-run a few times but no luck dropping to shell. Can you guide what else should i check ?

ERROR - tried to write to log file before it was open! The following won't be logged:
lvl2: 	EmulationStation - v2.0.0-rc1, built Dec 21 2014 - 20:33:47
ERROR - tried to write to log file before it was open! The following won't be logged:
lvl2: 	Creating surface...
ERROR - tried to write to log file before it was open! The following won't be logged:
lvl2: 	Created window successfully.
ERROR - tried to write to log file before it was open! The following won't be logged:
lvl2: 	Added unconfigured joystick Logitech Logitech Cordless RumblePad 2 (GUID: 030000006d04000019c2000011010000, instance ID: 0, device index: 0).
ERROR - tried to write to log file before it was open! The following won't be logged:
lvl2: 	Checking available OpenGL extensions...
ERROR - tried to write to log file before it was open! The following won't be logged:
lvl2: 	 ARB_texture_non_power_of_two: MISSING
ERROR - tried to write to log file before it was open! The following won't be logged:
lvl2: 	Loading system config file /etc/emulationstation/es_systems.cfg...
ERROR - tried to write to log file before it was open! The following won't be logged:
lvl2: 	Parsing XML file "/home/pi/.emulationstation/gamelists/esconfig/gamelist.xml"...
ERROR - tried to write to log file before it was open! The following won't be logged:
lvl1: 	System "atari2600" has no games! Ignoring it.
terminate called after throwing an instance of 'boost::filesystem3::filesystem_error'
  what():  boost::filesystem::create_directory: Permission denied: "/home/pi/.emulationstation/gamelists/atari2600"

adudney, 2015-01-02:
I had this same issue, turns out that my .emulationstation directory was registered as root:root rather than pi:pi, I used `sudo chown -hR pi:pi .emulationstation` to fix it.

mzeyrek, 2015-01-02:
i've changed permissions to everyone which wasnt recommended and it worked but not to pi, thanks for highlighting chown @adudney much appreciated, i believe your solution supposed to be the final fix for others.
@petrockblog i wonder if this is a known issue which supposed to be fixed with permission fix menu under retropie_setup or not.

ProxyCell, 2015-01-02:
i normally execute "sudo chown -R pi" for /home/pi - /opt/retropie and even /etc/emulationstation after my installs

i haven't noticed any issues with it yet

joolswills, 2015-01-02:
not sure how the directory got set to root, unless emulationstation was somehow run as root initially. It isn't a known issue afaik - and it's not happened when I have tested before with a vanilla raspbian. I will test again, could you give me the exact steps you did. Binaries install, followed by which set up options you ran etc. thanks.

mzeyrek, 2015-01-02:
I had a fresh install of latest raspbian, completed apt-get update && upgrade steps and finally upgraded firmware with rpi-update command. Rebooted and cloned repo as per instructed. I've got blue screen proceed for faster setup rather than compile, enabled autostart emulation under settings and performed reboot to get directly into emulationstation however got those errors.
Basically trick might be rpi-update ? I prefer to have everything updated before running the script.

adudney, 2015-01-02:
My steps:
Fresh raspbian install, setup wifi.
sudo aptitude update, sudo aptitude upgrade, sudo rpi-update.
 clone RetroPie, run setup.sh, select source installation and deselect most of it (Only install Nintendo consoles and emulation station with related configuration utilities), run (Takes a very long time), reboot.
Launch emulation station and it crashes with a boost error stating that it can't create some directory or another. (It differed between my installs, once it was GBA and the second time it was ScummVM)
If emulation station is set to run on launch and it crashes, the crash breaks physical keyboard input, I have not tried ssh yet.
Once setting the owner of .emulationstation and all subfiles to pi:pi rather than root:root, emulation station runs great, no crashes so far, even when launched on boot.

mzeyrek, 2015-01-02:
ssh possible, it exactly happened to me physical keyboard crashed (forgot to mention)  once recieved those errors and dropped into shell. I was unable to use my keyboard while have to connect via ssh for permission fix. I havent customized installation just was default

trentfaris242, 2015-03-09:
Was there ever a solution found for this, or at least a work around? I, too, am having this issue. The most frustrating part is that it crashes my keyboard so I'm limited to SSH.

adudney, 2015-03-10:
Set the owner of the .emulationstation folder and all subfiles to pi:pi rather than root:root

I used sudo chown -hR pi:pi .emulationstation

trentfaris242, 2015-03-10:
Oh I gotcha. I misunderstood the previous comments and thought it was the /etc/emulationstation/ directory. Thanks!

closed 2015-06-20 by joolswills

dethlab002, 2017-08-29:
how do you access permissions are the start up? my pi just blast through a boot screen, goes directly to ES and throws an instance which leaves me unable to do anything.