RetroPie/RetroPie-Setup #545: Serious issue with controller configs and hotkeys

ProxyCell, 2015-01-02:
Today I was modifying two controller configs on my retropie to have them contributed here and I noticed that in THIS repo directory: https://github.com/petrockblog/RetroPie-Setup/tree/master/supplementary/RetroArchConfigs

I found that most configs come with hotkeys already defined. Such as https://github.com/petrockblog/RetroPie-Setup/blob/master/supplementary/RetroArchConfigs/GreenAsia_Inc.____USB_Joystick_____.cfg

It defines the hotkeys as:
input_enable_hotkey_btn = "9"
input_exit_emulator_btn = "8"
input_menu_toggle_btn = "0"
input_load_state_btn = "6"
input_save_state_btn = "7"

Now I figured that I should add such hotkeys to my own controllers but this produced a VERY BAD result. I took a Logitech F710 and define the hotkeys:
input_enable_hotkey_btn = "8"
input_exit_emulator_btn = "7"
input_load_state_btn = "4"
input_save_state_btn = "5"
input_state_slot_increase_btn = "h0up"
input_state_slot_decrease_btn = "h0down"
input_menu_toggle_btn = "0"
input_rewind_btn = "2"
input_screenshot_btn = "3"
input_slowmotion_btn = "1"

This totally overwrote the function of those buttons. I tried it on a totally different controller and it also had the same terrible result. This means that instead of pressing the "enable hotkey button" AND START to quit out of the emulator, simply pressing START would now quit the emu... I have taken out such hotkey definitions but this means that by default this repo will provide people with controller configurations that simply do not work as intended.

joolswills, 2015-01-02:
You need to run the hotkey configuration setup I believe, to not have them enabled by default. Check the setup menu or use

```
sudo ./retropie_packages.sh hotkey
```

ProxyCell, 2015-01-02:
ok, but then don't the config files in this repo give people the hotkeys, overwriting the standard button functions of those controllers? should these not be removed or another solution?

joolswills, 2015-01-02:
I was assuming that people wanted to use the keys as hotkeys when pressing another key or something. Unsure as I'm a little unfamiliar with this part of the project. if you configure hotkeys to need alt pressed though, it shouldn't interfere with normal button presses - or are you saying it still does ?

ProxyCell, 2015-01-02:
the hotkeys apparently MUST be placed in the retroarch.cfg proper
please have a look at this thread:
http://blog.petrockblock.com/forums/topic/controller-cfg-saves-as-2axes11keysgamepad-cfg-cant-add-escape-combo/#post-84433

Same controller as mine and he has the same troubles. Placing the hotkeys into anything but the retroarch.cfg appears not to work.

My suggestion is to remove the hotkeys from the repo files here so that people's controller buttons work at least normally.

gizmo98, 2015-01-03:
Hi ProxyCell,

please try sudo ./retropie_packages.sh hotkey and select 2 or 3. This menu does not overwrite your hotkeys. It only sets the default keyboard behavior of hotkey_enable, exit and RGUI_menu key. Gamepad hotkeys like "select + hotkey" do only work if there is a keyboard binding for input_hotkey_enable.

See also:
https://github.com/petrockblog/RetroPie-Setup/pull/476
http://blog.petrockblock.com/forums/topic/default-configuration-for-retroarch-hotkey-button/

ProxyCell, 2015-01-03:
Hey Gizmo98, I think I will try to do a full recompile/reinstall down the line. I truly believe that having these hotkeys defined in the /RetroPie-Setup/tree/master/supplementary/RetroArchConfigs is a bad idea and overwrites the basic function of the buttons like START. now I could be wrong but aren't these config files later copied into the /opt/retropie/emulators/retroarch/configs ?

gizmo98, 2015-01-03:
Yeah. They are bad because RetroArch does not respect the input_enable_hotkey_btn (gamepad) if there is no input_enable_hotkey (keyboard). Some time ago the "escape" key was input_enable_hotkey and input_exit_emulator key and there was no problem.

ProxyCell, 2015-01-03:
well thats what i was talking about right? the fact that these files are used for autoconfig but will effectively make the start button only useful for exiting the emulator. isn't that a problem then?

petrockblog, 2015-01-03:
@gizmo98 
> Some time ago the "escape" key was input_enable_hotkey and input_exit_emulator key and there was no problem.

Could you post a pull request or maybe just a detailed comment that would fix this problem?

gizmo98, 2015-01-03:
In my opinion it is a retrorch issue. I will take a look into the coding.

gizmo98, 2015-01-03:
Ok. Retroarch has a function called check_block_hotkey() which checks if a hotkey enable key/button is set in tetrarch.cfg. It does not work if a hotkey_enable_button is set in an autoconfig file. I just have added autoconf input 0 (gamepad0) and it is working.
https://github.com/gizmo98/RetroArch/commit/4ae6a4149875bdeea38be5c173e641a2e94582eb#diff-380858814ac5b94a0f9ed4e4df5f344dR536

The only problem. If a autoconf gamepad is connected i must press "select" to use keyboard hotkeys. ;-)

gizmo98, 2015-01-04:
@ProxyCell please recompile RetroArch. RetroArch should respect the input_enable_hotkey set in a autoconfig file now.

Vendi76, 2018-06-05:
I have the same problem, with the same controller (logitech f710) using retropie 4.4 and retroarch 1.7.1. Trying to play psx games, but the controller act as if the buttons were used as hotkeys. Do You have any simple sollutions? Thanks