RetroPie/RetroPie-Setup #550: handleaptpackages module

joolswills, 2015-01-03:
We currently call this module for the binaries install, which removed some unneeded packages. However should a user be installing this on their main rasbian set up, they may well not want us to mess with their already installed packages.

My thoughts are to

a) rename this to something more explanatory with a good description - such as "Remove packages unneeded for RetroPie"
b) do not automatically call it from the binaries install, but just have it as an optional set up item
c) add some additional items such as "sonic-pi"
c) make a dialog with ok/cancel so users can decide if they really want to run it. (we could technically then still include it at the end the binaries set up, but allow users to choose)

Please let me know if you are happy for me to do this

petrockblog, 2015-01-03:
I vote for options b) and c) :-)

joolswills, 2015-01-03:
I seemed to have made two Cs - edited - please can you clarify. So you do not want it renamed then or description changed or ?

petrockblog, 2015-01-03:
I think, it would be good to:

a) rename this to something more explanatory with a good description - such as "Remove packages unneeded for RetroPie"
b) do not automatically call it from the binaries install, but just have it as an optional set up item
c) add some additional items such as "sonic-pi"

joolswills, 2015-01-03:
gotcha. Will do

joolswills, 2015-01-05:
done - https://github.com/petrockblog/RetroPie-Setup/commit/522dc60ba5708f9f5273490a38257f9c0f8983c1

Maybe we should remove the "-y" though so they can abort - I'm wary about the idea of removing stuff without letting the user decide first.

closed 2015-01-05 by joolswills