RetroPie/RetroPie-Setup #552: DOSBox error when installing from binaries

marcosps, 2015-01-03:
Hi guys,

Today I tried to setup the RetroPi in a fresh install of Raspbian, but I crashed in a problem with DOSBox. The error was about some files that where missing from /opt/retropi/dosbox, These files are missing from the binaries that are downloaded?

Thanks since now for this awesome project!

joolswills, 2015-01-03:
dosbox has changed since the last binary release, and another binary release is needed. Just ignore the error though, and if you need dosbox, you can install it from source via

```
sudo ./retropie_packages dosbox
```

marcosps, 2015-01-03:
Hi Jools,

thanks for the quick response. But now this error breaks the 
retropie_setup script by not creating the emulationstation bin/script. 
This can be avoided in the retropie_setup script?

joolswills, 2015-01-03:
can you be more specific in regards to what error etc. if the dosbox configure stage errors out on a binary install it should carry on configuring everything else.

joolswills, 2015-01-03:
aha. sorry I see the problem. My fault from a change I made earlirt. I'll fix that right now - you will need to update the RetroPie script. I'll let you know shortly when it's done.

marcosps, 2015-01-03:
That was my mistake. I should copy&paste the error here. Good to know 
that you're fixing the setup script.

I'll make a new test with the sources updated and tell you if the error 
is gone.

joolswills, 2015-01-03:
if you git pull now and re-run the install it should work (dosbox won't be installed but it won't fall over).

eddie, 2015-01-03:
Had this problem earlier, thanks @joolswills :+1:

joolswills, 2015-01-05:
New binaries are now released and the script is fixed up so I will close. Thanks

closed 2015-01-05 by joolswills