RetroPie/RetroPie-Setup #553: multiple binary archives idea

joolswills, 2015-01-03:
It takes a long to to build the binaries archive - even with distcc and parallel builds etc.

When stuff changes that adds new files, or changes the location (such as a the move from dosbox via deb to dosbox via source), people end up with breakage (recoverable by manually building missing stuff from source but still).

I was thinking it might be a nice idea to have multiple binary archives, one for each module (we can have a module flag for the ones that should produce a binary). We can then build individual archives as needed to upload and when something changes, rather than having to rebuild everything everytime. We would still want to rebuild everything sometimes to get up to date packages of course.

I'm pretty sure I can automate this in a nice way too. What do you think ?

petrockblog, 2015-01-03:
We could use this as an opportunity and start using the Raspbian packaging system. In a first step we can start generating packages for the individual modules, e.g., as described at http://ubuntuforums.org/showthread.php?t=910717. We should write a script that automates the creation and update of those packages as much as possible. These packages could be downloaded and installed via "dpkg -i PACKAGENAME". In a second step we could come up with a package repository that manages the packages such that the APT system of Raspbian could be used. But, again, as a first step the creation of the .deb packages without using a package repository would already be a huge step forward, I think.

joolswills, 2015-01-04:
I'd rather look at that after. Switching the system into split packages is an hours work or less. Creating debian packages is a few hours work per emulator at least. So happy to consider this as a start, but I have little motivation right now to build debian packages of everything :)

joolswills, 2015-01-04:
my 3 hours per module is probably way underestimated too. building debian packages takes serious time and effort.

joolswills, 2015-01-04:
checked that link - that is not a really sufficient example imho. Having built myself a variety of packages for ubuntu for Ubuntu on the O2 Joggler (http://joggler.exotica.org.uk/ubuntu/) we need to handle proper dependencies, build rules, and so on, and we need to be able to build source packages that can be signed for uploading to a repository etc. Just making some quick packages of some already made binaries isn't going to help with that really.

I'm not even sure if it is worth the effort the more I think about it (or at least I'm not particularly interested in spending time on it). the source and binary separation is a reasonable compromise imho. I'd rather spend time on the user interface / ease of use - (such as a the recent stuff to switch screenmodes, or the module improvements etc)

petrockblog, 2015-01-04:
I do not see why we need source packages. Sure, to use the full set of functionalities of the package system we would also have to consider that. But we could concentrate on the binaries and, hence, skip taking care for the other functionalities. 
We have the dependencies already considered in every single RetroPie module. Why can't we use these dependencies also in the individual packages?
With simple binaries archives we do not have a signature and no version information, but (we do not have to, but) we could have this with deb-packages.

Having said that I agree with you that even this surely takes more time than simply creating zip or tar.gz archives and I also think that we should concentrate on the ease of use for now: Let's go for the simple archives then.

joolswills, 2015-01-04:
I guess I was looking at the ultimate result of having a fully usable repository etc.

I see your point regarding building binary packages from the information we already have although it's not ideal in the way that the binary packages really should only have dependencies on the binary libraries rather than the development libs. I also am not convinced at the overall benefit of building some quick binary packages - we would still need our modules, source dependencies for building, configuration etc. However they would be simpler to build, and it might be scriptable in that case - if we are going to continue with our install location in /opt etc.

But ok, let's look at this after. You're the boss ;-)

petrockblog, 2015-01-04:
So, what should we do for now?
Option a) Leave the big binaries archive as it is now as it is
Option b) Split the big binaries archive into smaller ones

I thought that you proposed option b)?

joolswills, 2015-01-04:
Sorry, I was referring to looking at the packaging again later. I'm still all for splitting it up, as it gives a quick solution to being able to push out updates when something is broken etc.

petrockblog, 2015-01-04:
All right then - let's split the binaries archive up.

joolswills, 2015-01-04:
I'll branch off and submit a pull request once it is working for testing. thanks.

I am keen to push out a full binaries update before just to get things up to date - It's done but I need to do some testing to make sure everything is ok (I will compare against the last binary archive to make sure it is complete etc)

closed 2015-01-09 by petrockblog