RetroPie/RetroPie-Setup #556: Mupen64plus error during binaries-based installation

petrockblog, 2015-01-04:
Just observed this error during the installation of Mupen64plus with the new binaries archive:

```
= = = = = = = = = = = = = = = = = = = = =
Configuring N64 LibretroCore Mupen64Plus
= = = = = = = = = = = = = = = = = = = = =

Setting mupen64-gfxplugin = rice in /opt/retropie/configs/all/retroarch-core-options.cfg
Setting mupen64-gfxplugin-accuracy = low in /opt/retropie/configs/all/retroarch-core-options.cfg
Setting mupen64-screensize = 640x480 in /opt/retropie/configs/all/retroarch-core-options.cfg
cp: cannot stat `/opt/retropie/libretrocores/mupen64plus/data/mupen64plus.cht': No such file or directory
cp: cannot stat `/opt/retropie/libretrocores/mupen64plus/data/mupencheat.txt': No such file or directory
cp: cannot stat `/opt/retropie/libretrocores/mupen64plus/data/mupen64plus.ini': No such file or directory
cp: cannot stat `/opt/retropie/libretrocores/mupen64plus/data/font.ttf': No such file or directory
[esconfedit.py] Successfully saved to file /etc/emulationstation/es_systems.cfg
```

joolswills, 2015-01-04:
those files are in the new archive - I can only think somehow you are installing the old archive still or the new one didn't unpack ?

joolswills, 2015-01-04:
I've realised what's going on. It's in email - regarding switching over to the new archive.

petrockblog, 2015-01-04:
You are right - they are in the new archive.

closed 2015-01-04 by petrockblog