RetroPie/RetroPie-Setup #557: support for multiple platforms (ie non-RaspberryPi ARM boards)

spikegrobstein, 2015-01-05:
I've got a fork of this repo that I'm working with while trying to get many of the packages working on my ODROID XU3-Lite board. There are a handful of changes I've had to make to the package files in order to get things to either compile or to function correctly. I'd like to get these changes merged upstream at some point so that other people with different boards can use the software.

So I'm reaching out so we can figure out the best way to do this.

My thought is to set a variable somewhere for the `platform` and in each package, conditionally execute code (install packages, set flags, etc) depending on the value of that variable. I suppose the value could either be set as an environment var or set in a config or even guessed?

Some examples of differences I've come across:

 * my board doesn't have any `libraspberrypi-*` packages in `apt` to install as dependencies
 * RetroArch has a new `exynosdrm` dependency
 * `runcommand.sh` is not necessary
 * my `__default_cflags` in `retropie_packages.sh` is `-O2 -pipe -mfpu=neon -march=armv7-a -mfloat-abi=hard -marm`

Let me know if this makes sense or if you're interested in supporting alternative boards. I can put something together and send over a pull request, I just don't want to do a bunch of work that doesn't jive with your vision and have to redo it.

Thanks for the awesome project!

joolswills, 2015-01-06:
This will be up to @petrockblog to ultimately decide, but I see no problem with it so long as we can do it without too much logic etc. the stuff you have mentioned is quite doable. runcommand.sh can either be removed as part of the essystems generator function, or the runcommand.sh itself could just be a no-op on other platforms.

I'm rather tempted to get an odroid C1 too ;-)

spikegrobstein, 2015-01-06:
Cool, so give me the word (should @petrockblog be the guy to thumbs-up it?) and I'll put together a pull request and we can work out the details then. I haven't gotten every package working, yet, but i'll document which ones I know work and which ones I know need work. Perhaps in a `README-odroid.md` file?

I have to say, the odroid is amazing. the rpi is ok, but it definitely feels cheap and doesn't run so well, but the odroid is like a real computer; at least the XU3 is. so far, the emulators I've gotten working run amazingly well, even PSX. I highly recommend it.

joolswills, 2015-01-06:
for some packages that don't work we can have a platform flag or something so the system can automatically hide them.

ODROID XU3-Lite is nice, but 3x the price of a pi. I think the ODROID C1 will do very well - same price as a pi, but with more power. I really like the pi. It's a good price, and is very flexible, but it is starting to feel a little slow at times but I think it does better than my initial expectations. It serves its original purpose very well of course too.

petrockblog, 2015-01-06:
I would also be happy, if we could open the project to additional platforms. However, i totally agree with Jools that the logic for that support must be very simple: We should make sure that someone with a Raspberry Pi can develop, test, and and run a module with the need to have other hardware platforms. The same holds, of course, for other platforms as well. This should easy the further development of the scripts a lot.
Having some sort of a platform flag sounds like a reasonable way to start with. I am looking forward to the pull request(s) :-)

gizmo98, 2015-01-06:
I like the idea. I played around with the A20-OLinuXino (Allwinner A20) and a highly modified RetroPie-Setup worked well. SNES and PSX work better if there are no filters or higher resolutions. It was not possible to compile mupen64plus and the GPU driver seems to be slow and buggy under X. So there was no real benefit to switch to this platform.

joolswills, 2015-01-07:
I could implement some multi platform support into the main code if that would be useful to you - and we could go from there ?

petrockblog, 2015-01-07:
Sounds good to me.

joolswills, 2015-01-08:
https://github.com/petrockblog/RetroPie-Setup/pull/570

gabpaladino, 2015-01-13:
Hi guys, i'm starting with Odroid U3 to build a arcade.... RetroPie is awesome project to do it, other guys do some difficult or ugly systems...

How can i help on this goal? In my first test, have problem with FBA LibretroCore because can't install gcc4.8 on debian whezzy...

I used collabora repository from rpi to resolve libxkbcommon-dev dependence...

When finish i update if can play something.... kkk

gabpaladino, 2015-01-13:
@spikegrobstein your odroid-support branch is working fine? i see many changes on it.. 

maybe creating platform specific folders in scriptmodules subfolders are good option to avoiding many ifs on files...

joolswills, 2015-01-13:
having multiple scriptmodules for each platform would create a lot of duplication, however it is possible to do already as there are platform flags that can be set. so it would only be needed if the logical was not maintainable in a single module.

gabpaladino, 2015-01-13:
i think about shared modules in scriptmodules and specifics in plataform subfolder... want to help work on odroid u3, but I'm afraid of change and break the rpi

joolswills, 2015-01-13:
wait till @spikegrobstein has submitted some initial changests and then you can go from there perhaps ?

joolswills, 2015-07-13:
Closing this as no feedback - we have the initial framework for multiple platform support. I will look at getting the retroarch parts at least working on the odroid c1 at some point after 3.0/

closed 2015-07-13 by joolswills

spikegrobstein, 2015-07-13:
no worries. moving across the country totally knocked me out of this side project. thanks for everything!

juri74, 2015-12-09:
excuse me, i'm a bit lost, is retropie compatible with odroid xu4?

joolswills, 2015-12-09:
no sorry.