RetroPie/RetroPie-Setup #560: New UAE Emulator

njt1982, 2015-01-06:
I started playing with http://www.raspberrypi.org/forums/viewtopic.php?f=78&t=80602 last night and it seems like a much newer/better/actively maintained UAE emulator...

It also comes with a free kickstart so will at least boot SOMETHING (if not everything) without having to find your own kickstarts (beit legally or not).

I also started to rewrite the shell script that starts the UAE emulator so that it loads all the disks in a folder for a game and puts them as params to the emulator, eg `-s floppy0=CannonFodder1.adf -s floppy1=CannonFodder2.adf....`

I'll open a PR for that once I've had a chance to get all this working.

joolswills, 2015-01-06:
Funnily enough I posted there last night about incorporating that.

You are planning to do a PR for the emulator update also ? Sounds good. Note that I updated some uae4all stuff last night (fixing up the disk inserting script - you may want to check in regards to the runcommand stuff.

njt1982, 2015-01-06:
*boots up the pie and runs an update ;) *

njt1982, 2015-01-06:
Damn. Running update overwrite my script ;)

joolswills, 2015-01-06:
it should give a merge conflict, and you can compare/manually work out the changes etc.

joolswills, 2015-01-06:
(that's if you had conflicting changes of course)

njt1982, 2015-01-06:
nope - because I changes the `startAmigaDisk.sh` file in /opt directly, rather than in the scriptmodules :P

joolswills, 2015-01-06:
oops :/

njt1982, 2015-01-08:
Is there any documentation on writing the emulator installation scripts? `scriptmodules/emulators/uae4all.sh` ? I'm trying to understand how all the bits work...

njt1982, 2015-01-08:
Hmm

```
Could not successfully install Amiga emulator UAE4All (/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga, not found).

pi@raspberrypi ~/RetroPie-Setup $ stat /home/pi/RetroPie-Setup/tmp/build/uae4all/amiga
  File: `/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga'
  Size: 58        	Blocks: 8          IO Block: 4096   regular file
Device: b302h/45826d	Inode: 269401      Links: 1
Access: (0755/-rwxr-xr-x)  Uid: ( 1000/      pi)   Gid: ( 1000/      pi)
Access: 2015-01-08 21:36:33.736459736 +0000
Modify: 2014-12-01 04:10:37.000000000 +0000
Change: 2015-01-08 21:36:33.746459578 +0000
 Birth: -
```

joolswills, 2015-01-08:
no documentation current no - but there should be enough modules to work off as an example.

did the file exist at the point of the error? need more info - you can add set -x in your script to see exactly what is going on (and a lot of spam).

njt1982, 2015-01-08:
Ahh thanks - trying now...

njt1982, 2015-01-08:
So from debug I see this:

```
 = = = = = = = = = = = = = = = = = = = = =
Installing Amiga emulator UAE4All
= = = = = = = = = = = = = = = = = = = = =

+ install_uae4all
+ md_ret_files=('amiga', 'conf', 'data', 'kickstarts', 'lib', 'license.txt', 'readme.txt', 'saves', 'source', 'uae4all')
+ '[' '' '!=' '' ']'
+ '[' amiga, '!=' '' ']'
+ for file in '"${md_ret_files[@]}"'
+ '[' '!' -e /home/pi/RetroPie-Setup/tmp/build/uae4all/amiga, ']'
+ errors+=' Could not successfully install Amiga emulator UAE4All (/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga, not found).'
+ break
+ '[' -d /home/pi/RetroPie-Setup/tmp/build/uae4all ']'
+ find /home/pi/RetroPie-Setup/tmp/build/uae4all -maxdepth 0 -empty -exec rmdir '{}' ';'
+ '[' -d /opt/retropie/emulators/uae4all ']'
+ find /opt/retropie/emulators/uae4all -maxdepth 0 -empty -exec rmdir '{}' ';'
+ case "$mode" in
+ '[' 0 -ne 1 ']'
+ popd
/home/pi/RetroPie-Setup
+ '[' '!' -z ' Could not successfully install Amiga emulator UAE4All (/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga, not found).' ']'
+ __ERRMSGS+=' Could not successfully install Amiga emulator UAE4All (/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga, not found).'
+ return 1
+ return 1
+ [[ ! -z  Could not successfully install Amiga emulator UAE4All (/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga, not found). ]]
+ echo Could not successfully install Amiga emulator UAE4All '(/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga,' not 'found).'
Could not successfully install Amiga emulator UAE4All (/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga, not found).
+ [[ ! -z '' ]]
```

I'll add a "stat" just before the `-e` check...

njt1982, 2015-01-08:
`stat: cannot stat `/home/pi/RetroPie-Setup/tmp/build/uae4all/amiga,': No such file or directory`

Ahhh comma!!!

I'm too much of a PHP developer at heart ;-)

joolswills, 2015-01-08:
yeh - you are going to need a bash reference next to you to check as syntax is not always logical/common

joolswills, 2015-01-08:
md_ret_files is quite limiting currently, as it can not change the structure around when "installing" binaries. you can copy manually in that case. I'm probably going to change this at some point to something better. if you need to copy over say just  folder1/subfolder it would copy it to dest/ not to folder1/subfolder so in that case you would need to use cp.

njt1982, 2015-01-08:
Nah - the array method is fine here. I screwed up. In PHP/JavaScript (my *strength* languages), arrays are comma separated. Bash didn't complain, it just appended the comma to the path ;)

Now just need to sort out the configure part to regenerate the run script to work with game folders that use multiple disks. Currently the launcher links df0 to whatever disk you select.

My script (before I overwrote it by updating) listed all the disks in a folder and added them as params to uae as `-s floppy0=...`.

joolswills, 2015-01-08:
yeh. quoting is different. Just explaining that array in case as it's use is limited (I put it in as a stop-gap measure).

the emulator has built in disk switching support via a gui now right ? so maybe it's enough to insert a single disk, and the user can insert the rest as needed?

njt1982, 2015-01-08:
Yeah thats certainly a nice feature. It would be awesome if we could just select a game folder from ES and it launched the game outright. Or at least launched it with the disks selected.

It'd be nice to look for a conf file in the game folder and use that too... So each folder you put in ROMs could have a config file for disks, memory, cpu, graphics, etc, etc...

njt1982, 2015-01-08:
baby steps though ;)

njt1982, 2015-01-08:
Ah - damn... Cant use CLI to load disks in this version.