RetroPie/RetroPie-Setup #562: Backup and restore ROMS, BIOS, and CONFIG files

petrockblog, 2015-01-07:
The following idea [came from a user](http://blog.petrockblock.com/forums/topic/retropie-2-x-feedback-enhancements/#post-84754):

_[...] What may also be beneficial is an easy backup solution where only the main directories are backed up into a tar.gz or zip file. This file could be transferred to another ‘stock image’ and using the script, a restore option could be run which would populate all of the folders (roms, bios and configs) to the new installation based on the old._

KitchUK, 2015-01-14:
If this is possible it would make upgrading to new version ALOT easier, especially for non advanced users.

robertybob, 2015-01-14:
Images, bios, configs and gamelists I definitely agree with. Roms however I'm not sure about- what if a person has PS1 and SNES roms alongside NES and Genesis ..that's going to take a long long time to backup ..or you'll have to have an option to specify which roms too back up.

robertybob, 2015-02-03:
Bump. Any news on this @petrockblog ?

joolswills, 2015-03-21:
the usbromservice now has a facility to back up/restore the emulation station custom custom/gameslists etc More configs will be added later.

closed 2015-04-06 by petrockblog

Shawson, 2018-07-23:
Hey, did the config files element of this ever get going?  I use the usb rom service and my roms/ bios etc run from the hdd, but my configs folder is still on the sd card along with game save etc?