RetroPie/RetroPie-Setup #563: Add scripts/guide for creating the SD card image

petrockblog, 2015-01-07:
To make things reproducible and transparent to the community we must add the script(s) that were used for creating the latest SD card image.

ProxyCell, 2015-01-09:
so i run these on my own pi, and it'll create the .img for sharing based on how I have my pi configured?

joolswills, 2015-01-09:
they are for running on a pc to generate the distribution retropie image

petrockblog, 2015-01-11:
I have added a draft version for a HOW-TO to create the root filesystem with a RetroPie installation (see commit 5ab0f9927f5d741c363589961f4530c49b71698d). However, the draft still has quite a few open issues to be filled. 

Things that still have do be done are

 - Finish description of how to get a minimal Raspbian root filesystem
 - Description of changes to the system configuration (hostname, certain APT packages etc.)
 - Description of how to install RetroPie and with with options.
 - Description of how to generate the image archive

Still quite a few things, but we have already done every step above and it "only" needs to be described ...

joolswills, 2015-01-11:
nice!

the chroot.sh command i uploaded btw simplifies some of this - i think its still good to have it all detailed, but for convenience you can use it to chroot without having to remember to add/remove the qemu binary etc (and it sets the qemu arm cpu correctly to armv6l)

joolswills, 2016-08-11:
This can be closed now - all image building is included in retropie-setup via the "image" script. It's designed to be run on a PC  and requires QEMU.

```
# build rpi1 and rpi2 images
sudo ./retropie_packages.sh image all
```

closed 2016-08-11 by joolswills