RetroPie/RetroPie-Setup #564: RetroPie - USB gamepad working in emulation station but not in ROMS

RFKirb, 2015-01-07:
Firstly, thank you for taking the time to read my thread. Here's my current setup:

System: Raspberry Pi B+
SD: SanDisk Ultra 32GB
Distro: RetroPie v2.3
Gamepads: 2x SNES Retro USB Super Nintendo Controller (http://www.amazon.com/gp/product/B0034ZOAO0/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)

     As the title states, I can use either of my USB gamepads in emulation station just fine, but when I launch a ROM neither work, but the default keyboard bindings work. I followed the instructions here (https://github.com/petrockblog/RetroPie-Setup/wiki/Control-Issues) for USB gamepads, which listed the below commands:

$ /opt/retropie/emulators/RetroArch/installdir/bin/retroarch-joyconfig -a /opt/retropie/emulators/RetroArch/configs/gamepad.cfg

     The command executes fine, and I am able to use the tool to output the gamepad.cfg file, but again, upon entering a ROM (SNES or NES) both gamepads are unresponsive. What should be my next step? I appreciate your help.

ProxyCell, 2015-01-08:
ok, you're using a fresh 2.3 image right? have you ran the "update scripts"?
i have multiples of this EXACT controller. i just ran the "Register RetroArch Controller" to get the correct config for it originally

RFKirb, 2015-01-08:
Yup, fresh 2.3 image. I'll try that when I get home.

ProxyCell, 2015-01-08:
let us know the results when you do

gizmo98, 2015-01-08:
The current folder names are:
$ /opt/retropie/emulators/retroarch/tools/retroarch-joyconfig -a /opt/retropie/emulators/retroarch/configs/gamepad.cfg

instead of

$ /opt/retropie/emulators/RetroArch/installdir/bin/retroarch-joyconfig -a /opt/retropie/emulators/RetroArch/configs/gamepad.cfg

ProxyCell, 2015-01-08:
@gizmo98 
for 2.3?

gizmo98, 2015-01-08:
Current! If you update RetroPie-Setup or if you use Retropie 2.4 Beta ;-)

RFKirb, 2015-01-08:
So I updated the binaries and scripts, then configured the configure retroarch controller from the retropie setup, and it worked. Can I use it again to configure a second controller? Thanks everyone for your help.

ProxyCell, 2015-01-09:
yeah i guess but is that 2nd controller the same?? if it is then you don't have to do anything extra

RFKirb, 2015-01-09:
Yeah, but the .cfg file splits it out by controller #, so you can
potentially use different types (SNES, N64, etc). That being said, I know I
can copy the text from controller #0 and paste it to controller #1, but
using the utility would be easier.

lomiccw1, 2015-01-09:
I had this issue and had to copy the .cfg file generated in $ /opt/retropie/emulators/retroarch/configs to $ /opt/retropie/emulators/RetroArch/configs  - why the split? I don't get the difference. I understand the directories are case sensitive, but what is the diff between RetroArch & retroarch?

joolswills, 2015-01-09:
/opt/retropie/emulators/RetroArch/configs is the old location. Your install is out of date.

joolswills, 2015-01-09:
Recommend you download the 2.4 beta image and go from there

joolswills, 2015-01-09:
(alternatively, you should remove /opt/retropie completely and do a fresh binaries install from the latest setup script)

ProxyCell, 2015-01-09:
hey @joolswills have you posted it elsewhere?

joolswills, 2015-01-09:
it is available from http://blog.petrockblock.com/retropie/retropie-downloads/

lomiccw1, 2015-01-09:
Awesome! thank you so much.

lomiccw1, 2015-01-09:
Sorry, this might not be the correct spot to ask, but: are the install binaries still updated? I have seen info that indicated that the  pre-compiled binaries were no longer supported by Github. 

https://github.com/petrockblog/RetroPie-Setup

"Binaries and Sources:
RetroPie Setup offered the possibility to only install RetroArch, the cores, EmulationStation, and SNESDev with pre-compiled binaries. Since Github has discontinued the download service the option for installing pre-compiled binaried with the RetroPie script was discarded. Take your time and let RetroPie Setup download and compile the programs from their sources!"

joolswills, 2015-01-09:
It is the wrong place to ask - open a new ticket or use the forum - or else issues will get very confusing. the binaries are updated.

GuGuss, 2015-03-20:
Hi,

I'm having the same issue here with exactly the same controllers/Pi. I can use the gamepads in the EmulationStation, but not in any roms (only the keyboard works).

I've installed the image that I downloaded from here: http://blog.petrockblock.com/retropie/retropie-downloads/download-info/retropie-sd-card-image-for-rpi-version-1/

Any idea?

ProxyCell, 2015-03-20:
hmm, can you go into the source-build menu? disable everything except for the "retroarchautoconf - RetroArch-AutoConfigs" near the bottom.

that should install the controller configs for you

closed 2015-04-06 by petrockblog

ias2, 2016-08-09:
Hi, I've the same problem. The only way to get the joystick working is to connect the ethernet cable to raspberry (!). What is the relationship between the joystick and the lan?

joolswills, 2016-08-09:
This issue is closed - please use the forum for support.