RetroPie/RetroPie-Setup #566: Double rom folders

ProxyCell, 2015-01-08:
This is a query about some "double rom folders"
fba and fba-libretro
pcengine and pcengine-libretro (both are libretro)

Is the intention to do this for other emulators down the line so people can pick their own emulator but have double the roms?

While this isn't a terrible idea there is currently no way for the user in Emulatation Station to determine which emulator is which. Could this be done with themes?

If this is not the intention then are these to be merged like other how every other system is currently handled, one roms folder with the es_systems.cfg managing who loads the roms.

I have attached a screenshot of roms directory

[image]

joolswills, 2015-01-08:
it is so people can use one emulator or another for stuff yeh.

you are right it's confusing on emulationstation - I raised this upstream, and was given a solution, but it requires customisation of the theme to do some duplication and some additional includes. It is something I plan to look into.

See here for details

https://github.com/Aloshi/EmulationStation/issues/317

ProxyCell, 2015-01-08:
thanks joolswills - i look forward to seeing that develop
what about the roms themselves? is it possible to have them placed into one single roms folder, then pick the emulator you prefer by switching to the differently themed group in ES? it would cut disk space usage in half while you use multiple emulators

joolswills, 2015-01-08:
no, but you can cut disk usage by symlinking folders if you want to keep complete rom sets in each folder.

closed 2015-01-08 by ProxyCell