RetroPie/RetroPie-Setup #569: style guide

joolswills, 2015-01-08:
since we don't have one, perhaps we should decide on some stuff ?

What shall we do

```
function name() {

}
or
function name()
{

}
```

right now we mix both, but we mostly use the first form in modules, and the 2nd in helpers.sh etc.

string / variable matching

```
[ "$var" = "value" ] (generic shell)

[ "$var" == "value" ] (bash / shell specific )

[[ "$var" == "value" ]] (bash specific)
```

we used a mix currently. if we ever wanted our script to not be bash specific we would have to use the first case I guess. We use some bashisms around the code though, so maybe it doesn't matter - but we should stick to one I guess.

function names - most helper functions are camelCase - the rest is "something_something". we can't really change the modules so im not too fussed about this - it's a lot of search replace to fix up the helpers if we wanted, but possible. 

variable names - for most functions we use short words, without underscores or case changes. seems mostly consistent.

global variable names - we have a couple of uppercase __ERRMSG/__INFOMSG ones, the rest are lowercase. stuff that is closer to a "constant" is usually __variable - like freemem etc but things like $rootdir / $scriptdir are not. I guess it's not too bad.

indentation - 4 spaces.. this is consistent across the code now.

petrockblog, 2015-01-11:
You are right - until now we did not decide on a style guide.

I have done some browsing, found, e.g., the Google Shell Style Guide and adapted it a little bit. I have added the @RetroPie-Setup Script Style Guide" to the Wiki at https://github.com/petrockblog/RetroPie-Setup/wiki/Shell-Style-Guide.

It addresses your points from above and also some additional ones. What do you think?

joolswills, 2015-01-11:
yeh looks good. I found a quoting issue due to markup I fixed, there might be more.

currently we don't have camel case for some vars, but mostly they are single words, so it's fine. A few cases they are twowords rather than twoWords.

We can fix up any [ ] to [[ ]] as we find them. Probably most from me - old habits :)

joolswills, 2015-01-11:
we currently have a bunch of global __vars - I dont think it's a problem to keep their naming, but we should make them read-only once initially set.

in that styleguide it says to use camelCase variables, but then it shows examples with zip_version = perhaps we can remove the camelCase variable requirement - i don't see a problem with our current variables really and we have plenty with _ in too. What do you think ?

petrockblog, 2015-01-11:
Uups, that was an error. It is ok for me if we allow both camel case and _-notation. I will adapt the guide.

joolswills, 2015-01-11:
I dont think we do use any camelCase variablenames currently so maybe it's ok we just keep it as something or something_something and have __something for read only global constants ?

joolswills, 2015-01-11:
I like the camelCase helper function names though etc as you laid it out. makes sense (and matches what we do currently)

joolswills, 2015-01-11:
I'm currently converting all the [ ] to [[]] - ill submit a pull request rather than committing directly so we can have more eyes to spot any errors.

petrockblog, 2015-01-11:
All right!

petrockblog, 2015-01-11:
BTW, I have just found http://www.shellcheck.net, a shell script linter that can also be installed locally. Might be interesting to have a look at ...

joolswills, 2015-01-11:
thanks installed

joolswills, 2015-01-11:
I changed the styleguide so it specified in examples "==" and not "=" - Since it first mentioned == it's probably a mistake it later has = and I think we should stick to one for conformity.

petrockblog, 2015-01-11:
Ok

petrockblog, 2015-03-04:
The style guide can be found at https://github.com/petrockblog/RetroPie-Setup/wiki/Shell-Style-Guide now.

closed 2015-03-04 by petrockblog