RetroPie/RetroPie-Setup #571: Script broken since recent update

musyne, 2015-01-09:
Hi,
Yesterday it worked fine.
Today when I run a normal binary install on a fresh raspbian, EmulationStation says that no system are configured.

joolswills, 2015-01-09:
did you update the script today ?

please try running the binaries install, and uploading the install log somewhere for me. thanks.

joolswills, 2015-01-09:
(please update the script first)

joolswills, 2015-01-09:
(in the meantime, I will try and replicate the problem on a fresh install)

joolswills, 2015-01-09:
I found a major problem that happened due to the order of a few lines of code that got changed (by me). Yes, I broke it.

Please update the script, and try again and it should set things up correctly.

joolswills, 2015-01-10:
Is it working ok for you again ?

musyne, 2015-01-11:
Yes.

closed 2015-01-11 by musyne