RetroPie/RetroPie-Setup #574: Configure PRDoom to support midi files in WAD

stsfin, 2015-01-09:
Please configure PRDoom to support the included midifiles in wads, this saves space installs etc and removes the hastle in converting music to MP3

PRDoom does support the Midi music, but needs the Timidity instrument patches from the SDL_mixer

"The SDL_mixer library used by PrBoom supports software MIDI music synthesis. If you want to hear the Doom music, you need a set of Timidity instrument patches. Get the timidity.tar.gz from the bottom of the SDL_mixer page, and extract in /usr/local/lib. You may need to symlink /usr/local/lib/timidity/timidity.cfg to /etc/timidity” see the console output from PrBoom to see where it is looking for this file."

joolswills, 2015-01-09:
We actually already install the timidity files as part of the prboom configuration. You are right it doesn't seem to be using them though,but  are we sure the libretro version has support though? - this is not prdoom we have but libretro-prboom which is a port to libretro of prdoom. I will look into it however.

joolswills, 2015-01-09:
I meant prboom not prdoom. There are mentioned in the libretro-prdoom issues that the libretro version doesn't have midi music so it may be the case you need to use mp3s.

https://github.com/libretro/libretro-prboom/issues/1#issuecomment-5804201

joolswills, 2015-01-09:
https://github.com/petrockblog/RetroPie-Setup/wiki/PRBoom---Doom-Emulator

joolswills, 2015-01-09:
@gizmo98 

The timidity stuff was added in c99ed9643ccb8df00c33a88bcaa98456a73f3705 - can you confirm if the libretro prboom port supports midi or not ? Oddly the commit makes no reference to this change at all.

gizmo98, 2015-01-10:
I had the same thought as stsfin. I have no midi sound with prboom-libretro, tyrquake-libretro and so on. So it could be that the software midi lib timidity only works with standalone emus.

stsfin, 2015-01-10:
@joolswills fast google reveals that prboom-retrocore has the sdlmidi support: https://github.com/libretro/libretro-prboom/blob/master/README but im not sure if its in the windows version only

stsfin, 2015-01-10:
Actually I think I know the culprit why it is not working, timidy is installed in the beta build but there are no midi patch files, I'll try this today and report later.

gizmo98, 2015-01-10:
You can try to install "apt-get install timidity" and "apt-get install fluid-soundfont-gm".

joolswills, 2015-01-10:
that readme is from prboom - unchanged for the libretro port.

joolswills, 2015-01-10:
It's unlikely the libretro core has support, as it would have to re-implement it itself in a way that would work without requiring system requisites like timidity

joolswills, 2015-01-10:
Post a ticket/question up on the libretro-prboom github site :)

stsfin, 2015-01-10:
Will close this issue, since its not RetroPie related. Solution was just to rip the Mid files from the wad and convert them to mp3, now i have the "real" music, not the PSX ones and its good enough for me.

closed 2015-01-10 by stsfin

ViperAcidZX, 2020-07-06:
Sorry to necro an old issue thread, but is there a way to get PrBoom+ (source build) to work with different soundfonts like EmperorGrieferus' SC-55 soundfont?