RetroPie/RetroPie-Setup #578: Add support for an automatic scraper

sselph, 2015-01-11:
Hi,

I am building a scraper https://github.com/sselph/scraper for console roms that scrapes them automatically. It does it by hashing the rom data and looking it up in a DB of hash to thegamesdb ID. The goal being to get no false positives and with a high hit rate. It has been suggested I reach out to see if this is something that could be integrated into retropie's scripts to help new users get a good baseline gamelist.xml.

If y'all are interested I can add some tweaks to make it more aware it is running in a retropie installation and scrape all supported systems, placing gamelists and images in the correct folders, etc.

Let me know if this is something you want to try and integrate and any other ideas on how it would need to function to be useful.

joolswills, 2015-01-12:
Sounds good to me :) perhaps it can be integrated as a supplementary scriptmodule. What do you think @petrockblog ?

robertybob, 2015-01-14:
I've used this scraper, albeit on Windows, to scrape my Mega Drive and Snes collection and I calculate it's only missed around 25 out of 1250 roms and inaccurately scraped less than a dozen more. It's A LOT quicker than letting Emulation scrape over a thousand roms and I highly recommend it :)

petrockblog, 2015-01-14:
Integrating it as a module sounds good to me. We can wait for further feedback from the users then to decide whether to keep it or not.

Can the script be integrated into ES? I have not looked into those details yet.

sselph, 2015-01-15:
My original plan was to try and add it as a supplementary module where the user selects to update/install the scraper or select the platforms to scrape. I haven't tried anything yet I wanted to make sure it would be a welcomed change before spending too much time on it.

ES could do similar logic if it wanted, triggering it as a separate binary. It could just run the scraper or I could add a few flags so it could call it like scraper --get_gdb_id path/to/rom and could use its logic from there to build the gamelist, falling back to searching and asking the user if I can't provide an id. It is also technically possible for the code to be compiled and linked into the c code but I don't have any experience with that.

Aloshi, 2015-01-15:
I might take a look into integrating this with ES directly if that is in fact possible.  I don't know any Go, though, and it would also require adding a Go compiler to the ES dependencies/build process.

sselph, 2015-01-15:
The main bit of code that isn't already in ES is the code to hash different rom formats. The Decode functions take a file reader and return a new reader that returns the rom data without headers that hasn't been byte swapped. That data is sha1 hashed and then looked up in a mapping imported from a csv. If compiling that Go code to a static library proves too difficult I can try and learn enough C to rewrite those functions.

Aloshi, 2015-01-15:
Hmm, okay. I always wanted ES to have CRC-based scraping, but none of the online databases seemed to support it. You can actually find a reference to this in the ScraperSearchComponent code, one of the enum values for search type is "ALWAYS_ACCEPT_MATCHING_CRC" or something.

sselph, 2015-01-16:
I went ahead and played around with converting the hashing code to C++. I created a shasum-like utility in c++ using the new code. I haven't the ability to inspect inside zip files to see if a file matches a rom extension and hash that. The code is ugly and needs cleaning up. I guess I can move this bug over to ES and work on submitting a pull request there.
https://github.com/sselph/scraper/tree/master/cpp

closed 2015-01-17 by sselph