RetroPie/RetroPie-Setup #583: BBC Micro emulators

diegop77, 2015-01-12:
This is the link of the project that have a version for linux of the BBC- Micro computer, can you add this on RetroPie?

Link: http://b-em.bbcmicro.com/

Thank you.

Diego

alexxxmg, 2015-12-29:
Having just installed RetroPie - and having a lot of fun with it - I was surprised not finding the emulation of the BBC (just considering some really obscure systems it is able to emulate!)
I cast my vote for it being supported!!!

objectchris, 2016-01-07:
looks like the full installation process is detailed here http://haprpi.blogspot.co.uk/p/the-bbc-lives.html?m=1 - looks like it should be possible!

alexxxmg, 2016-01-07:
thank you objectchris!

oneswitch, 2016-06-09:
Agreed. Would love for there to be an easy default installation of a quality BBC Micro emulator for RetroPie. That would be fantastic for many reasons. The best emulator on Windows I've found is BeebEm: http://www.mkw.me.uk/beebem/ - and there's a UNIX/Linuz version here: http://beebem-unix.bbcmicro.com/

Marshallx, 2016-11-20:
wow bbc micro not included by default, what gives? +1

linky00, 2017-01-04:
Jesus man, I wanna play Elite! I'd love to see a default BBC Micro emulator!

methanoid, 2018-11-17:
Periodically I google to see if this exists.... Does nobody care about the BBC Micro for emulation? I just assumed Retropie would have it! :-(

nmstoker, 2018-12-19:
The instructions linked to by @dankcushions work well (http://haprpi.blogspot.co.uk/p/the-bbc-lives.html?m=1 ) it just needs a bit of polish
Haven't got sound working yet, but will try that over the holidays