RetroPie/RetroPie-Setup #585: Emulators not returning to ES

dudleydes, 2015-01-12:
A number of non-Retroarch emulators fail to return Emulationstation. I am presented with a black screen and the pi freezes. I can SSH into the pi but sudo reboot doesn’t work. The only way to fix this is through a power cycle.

There is an open issue at the emulationstation github (https://github.com/Aloshi/EmulationStation/issues/206). In the last post, static-inc has posted a solution. His suggestion is to change line 274 in runcommand.sh from

eval $command

to

eval $command &
wait $!

I have tested it for jzintv. Everything works as it should. Would it possible to include this change?

joolswills, 2015-01-12:
I made some comments over at the other thread - https://github.com/Aloshi/EmulationStation/issues/206#issuecomment-69654177 

I'm not sure why the above would fix it, as I can't see why it could give any difference behaviour from what we do now - unless jzintv automatically backgrounds itself - I suspect the issue is more to do with resetting the framebuffer after an emulator leaves it in a unusable state

joolswills, 2015-01-12:
I have committed a change to runcommand in https://github.com/petrockblog/RetroPie-Setup/commit/0e19b4c6fa45bdf681b60f2329b410954239d8ad I will try and reproduce the problem also. I am not keen on adding the other change without a bit more information as to the workings.

dudleydes, 2015-01-12:
I have tested the new version of the runcommand script and jzintv returns to ES without any issue. I haven't used any of the other emulators mentioned such as vice or fuse-sdl.

joolswills, 2015-01-12:
thanks for the feedback.

static-int, 2015-01-14:
To clarify my last comment in [Aloshi/EmulationStation#206](https://github.com/Aloshi/EmulationStation/issues/206)..

The ```eval $command``` never returns to continue executing subsequent statements from *runcommand.sh* so the ```if [ $switched -eq 1 ]; then...```.

static-int, 2015-01-14:
To clarify my last last comment in [Alosh/EmulationStation#206](https://github.com/Aloshi/EmulationStation/issues/206)..

The `eval $command` when run in the foreground is not returning and as a result the subsequent statements in **runcommand.sh** (and hence the `if [ $switched -eq 1 ]; then...`) are never getting called.

For some reason the eval when launched in the foreground is missing the child signalling it's exit (x64 in my case) and so never returns from it's implicit wait loop.  And without the underlying call to wait() to reap the child exit status from the proces table the child stays behind as a zombie/defunct process. 

Forcing the subshell to return immediately with `eval $command &` however, and calling `wait $!` explicitley for the most recently executed background process reaps the child as expected.

On a side note x64 is not crashing for me and inspecting `$?` after the call to `wait $!` returns zero.

joolswills, 2015-01-14:
ok. I was unable to reproduce this however - I am unable to make x64 leave a zombie/defunct process but when I tested it with some test code which does create a defunct process, both cases behaved the same. the wait waited for the child process, so never returned also. 

it;s a situation that shouldn't happen anyway - but may not be needed with the latest retropie-setup as I have done some work on the vice emulator.

you can update by updating retropie-setup and then doing

```
sudo ./retropie_packages vice install_bin
sudo ./retropie_packages vice configure
```

Sorry if I sound resistant to the change, I just like to be able to reproduce the issues first, and try to fix up the faults, and only put in a workaround if absolutely necessary.

static-int, 2015-01-14:
I have updated vice as mentioned and can confirm that my workaround is no longer necessary (at least with vice, I have not tried others).  I do have separate config files in ~/.emulationstation and ~/.vice.  My SDLBitdepth is also 8, but my SoundDeviceName is set to sdl.

Also, no appology necessary.  Your responses were well received and I appreciate your hard work.  Thank you!

Pintglass, 2015-01-14:
I have updated vice  as suggested above but I still get a black screen on exit, I can ssh in to the pi but reboot or poweroff commands have no effect.

Thanks Pintglass

joolswills, 2015-01-14:
did you also do

```
sudo ./retropie_packages runcommand configure
```

and did you update the script first also ? vice should use dispmanx as default - but may not if you have an old runcommand installed.

Pintglass, 2015-01-14:
I hadn't done the sudo ./retropie_packages runcommand configure command, but have just done it after you posted but unfortunately I still get the black screen on exiting vice, I had updated the script first then ran you first 2 commands above. When I ran the last command you posted it didn't return anything in terminal is that correct?

joolswills, 2015-01-14:
it should have printed it was configuring "runcommand".. I made a typo btw - it should be ./retropie_packages.sh rather than ./retropie_packages  - doing a full binary install is easiest - depending on your level of linux shell knowledge etc.

joolswills, 2015-01-14:
I initially wrote "runcommand configure" which was a mistake and would have caused no output.

just 

```
sudo ./retropie_packages.sh runcommand
```

should install it.

Pintglass, 2015-01-15:
I have tried the above command, and get confirmation from it but I still get the blank screen on exiting vice.

 Just to recap first I updated retropie-setup from the menu then ran the three commands as posted
sudo . /retropie_packages.sh vice install_bin
sudo . /retropie_packages.sh vice configure
Sudo . /retropie_packages.sh runcommand

joolswills, 2015-01-15:
that should update everything. what does it say for vice when you do 

```
sudo . /retropie_packages.sh dispmanx
```

try changing the setting and seeing if that helps.

Pintglass, 2015-01-15:
After running the above command vice is the only emulator in the list that is enabled, is this the correct setting?

joolswills, 2015-01-15:
yes, unless you update the other emulators - where there will be some more enabled by default.

joolswills, 2015-01-15:
you may also try to switch screenmode on starting the emulator (pressing m or x to bring up a menu).

joolswills, 2015-01-15:
what kernel version are you running btw ?

Pintglass, 2015-01-15:
Linux raspberrypi 3.12.26-rpi-aufs #1 PREEMPT Thu Aug 14 10:48:01 CEST 2014 armv6l GNU/Linux

joolswills, 2015-01-15:
you may want to apt-get update; apt-get dist-ugprade (or just run the full binaries install - you probably have plenty of other things out of date also)

Pintglass, 2015-01-15:
OK I'II update, as I just tried setting the screenmode on launch to cea1 640X480 but I Istill get the blank screen.

joolswills, 2015-01-15:
btw you got a black screen after disabling dispmanx sdl for vice also ?

Pintglass, 2015-01-15:
Sorry no that was with the dispmanx sdl setting set to enabled, just updating at the min, so will try with it disabled when its updated

Pintglass, 2015-01-15:
I tried it with the dispmanx sdl setting set to disable but again I get the same problem of blank screen
That was after apt-get dist-upgrade

joolswills, 2015-01-16:
weird. I am unable to reproduce the problem here. What video output are you using (port / resolution). 

To rule out something different with your install that is causing a problem please try this image - https://drive.google.com/file/d/0B_knGioK16E6RFYyVW1iNld5aFU/view?usp=sharing

joolswills, 2015-01-16:
thinking about it, you may even lack the sdl dispmanx libraries if you haven't updated recently - the initial instructions on this thread were really designed for just updated vice on a recent image. if your image base is retropie 2.3 it will need a full binaries update first (but is worth removing /opt/retropie as all the emulator locations changed etc) So Let's start with the new image above and go from there.

Pintglass, 2015-01-16:
Im using the HDMI port with a resolution of 1920 x 1080, I will try the image from above tomorrow.

Pintglass, 2015-01-17:
After installing from the image posted above I no longer get the blank screen I'm returned to emulationstation after exiting vice, I have not tested with any other emulators.

I assume  updating Binaries and doing apt-get dist-upgrade is not sufficient to fix the error, is installing the above image the only solution?

Thanks Pintglass

joolswills, 2015-01-17:
you probably lacked the dispmanx sdl at it's new location so it wasn't being used. Updating all binaries is sufficient usually, but updating just the ones mentioned earlier is not if it had been a long time since you had updated everything. A lot has changed since retropie 2.3 image, so I would recommend using the new image - it's also a lot slimmer than the previous image.

Pintglass, 2015-01-18:
I know this is the wrong place, but I'm not sure if I should open a new issue, but with the above image the zxspectrum emulator will not load the rom selected in emulationstation, also I have my rpi connected to an amp then to my TV for some reason my amp is detecting a DTS signal from emulationstation.
Could you let me know if I should open a new issue for this of if I should post elsewhere?

joolswills, 2015-01-18:
a new issue - they are unrelated

joolswills, 2015-01-18:
Before opening an issue I would get help on the forum first though.

closed 2015-01-18 by joolswills