RetroPie/RetroPie-Setup #586: Could not initialize sdl no available video device

expandables2, 2015-01-13:
Hi I booted to tty1  when i login i am getting could not initialize sdl no available video device. But when i  boot  to tty7 EmulationStation shows up . I have a menu item with a bash script that says startx and when i startx it loads a desktop session on tty8 and it doesn't work as well because the sound is off  Any help on whats going on? I install Emulationstation manually. I am guessing that opengl is not being loaded, how do i make it load before anything else so EmulationStation will start.

joolswills, 2015-01-13:
how did you install ? you need to provide more information thanks.

expandables2, 2015-01-13:
I install on raspian Installing on Raspberry Pi (Stand-alone)
http://emulationstation.org/gettingstarted.html#install_rpi_standalone

joolswills, 2015-01-13:
you are installing emulationstation / SDL2 from source manually ? you probably need to add "--disable-video-opengl" to sdl2 configure - but you are posting in th wrong place.

This issue tracker is for retropie - you do not need to do anything manually like that for retropie and emulationstation will work out of the box.

expandables2, 2015-01-13:
Hi I appreciate the help and understand this isn't the right place to ask but just wanted some help thanks . One last help where can i add "--disable-video-opengl" to sdl2 configure?

joolswills, 2015-01-13:
to the "./configure" line. But why not just use retropie to build emualtionstation ?

expandables2, 2015-01-13:
I have retropie already s amazing but I wanted to learn couple stuff so i install emulationstation manually on another sd card

expandables2, 2015-01-13:
so do i sudo nano configure and add --disable-video-opengl

joolswills, 2015-01-13:
no. in the instructions where it says to run ./configure for sdl2 you should add that additional parameter. 

I recommend using a forum if you need further help - this is out of the scope of our issue tracker.

closed 2015-01-13 by joolswills

expandables2, 2015-01-13:
Hey one last request please? When i exit emulationstation it goes to tty7 x windows how did you disable tty7 and x windows  to get it to exit into a terminal

joolswills, 2015-01-13:
we don't run it in x windows at all. we run it from the console.

expandables2, 2015-01-13:
I ran it from tty1 and when i exit EmulationStation it goes to tty7 instead of staying in tty1 terminal. So do i need to disable tty7 if so how can i do that? I really appreciate the help thanks

joolswills, 2015-01-13:
I don't know. I recommend you find a suitable forum for further help.

expandables2, 2015-01-13:
Hi and you was able to disable X? I try a ton of forums got no answer please.

expandables2, 2015-01-13:
When I exit the EMS it takes me to tty7 x windows. I have the EMS in tty1 why does it take me to tty7?  Or how can i disable x windows and tty7?