RetroPie/RetroPie-Setup #589: Add improved ScummVM

KitchUK, 2015-01-13:
Details here: http://www.raspberrypi.org/forums/viewtopic.php?f=78&t=96258&sid=c69bae3e7481c681293733f972947f11

joolswills, 2015-01-13:
thanks for the heads up. It is now included.

joolswills, 2015-01-14:
erk. works on composite but gives a black screen on hdmi out - need to look into it further.

joolswills, 2015-01-14:
I have reverted this change for now - needs more testing. Might be worth us also looking at the libretro scummvm core

petrockblog, 2015-04-06:
Can this issue be closed?

joolswills, 2015-04-06:
I still have some planned stuff with this, so it can be left open for a bit more. Hoping to get some more time to test vanfanel's newer dispmanx backend (which he did after the egl + dispmanx one)

HerbFargus, 2015-04-06:
This is semi-related, but some users have had issues with the hardcoded controls in Scummvm,  Would adding the libretro scummvm core give more flexibility when it comes to controller configurations?

joolswills, 2015-04-06:
I would certainly like to add the libretro core as an option (which is much more viable now we have the emulator switching functionality)

HerbFargus, 2016-02-29:
One thing that might be worth considering is the `--enable-all-engines` flag. This provides some extra engines for more experimental games like Myst and Riven. may also be worth looking at compiling from master rather than a point release tarball.

joolswills, 2016-02-29:
I am waiting on the 1.8 release - then I will update the module to build using rpi backend. being able to build from stable / bleeding edge will be part of the more modular stuff I have in mind, but our shipped binaries will likely be kept at the last stable.

BenWlson, 2016-03-06:
Just a heads up, Version 1.8 was officially released March 4th.

joolswills, 2016-03-06:
Thanks. I will update it (with rpi backend)

joolswills, 2016-03-26:
This is done.

closed 2016-03-26 by joolswills