RetroPie/RetroPie-Setup #591: CATSFC SNES emulator

KitchUK, 2015-01-14:
Should this libretro core be added to Retropie?  https://github.com/libretro/CATSFC-libretro

joolswills, 2015-01-14:
I don't know - should it ? How does it compare to the other snes cores ? :)

ProxyCell, 2015-01-14:
allegedly it is quite good and modern. it is a fork of snes9x and is optimized for low end devices
"Raspberry Pi users and other devices like that should already be reaping the fruits of the labor being done right now." - http://www.libretro.com/index.php/in-the-run-up-to-retroarch-1-1-whats-new-pt-3/

joolswills, 2015-01-14:
ok sounds good. We could add it as experimental and go from there perhaps. We also have armsnes and pocketsnes - I wonder if we will get to a point where we don't need all of them ? :)

ProxyCell, 2015-01-14:
much like the FBA/PCE stuff I believe that keeping them all is for the best. armsnes has great performance but terrible compatibility, etc, etc. you floated an idea about letting people switch emulators somehow, i think that'll be the biggest deal down the line

joolswills, 2015-01-14:
well, ideally we would have one perfect emulator for each platform :) - the more emulators there are, the more there is to maintain, configure, etc.

ProxyCell, 2015-01-15:
thats true... but i think deep down you know that the only way for there to be "one perfect emulator" is for there to be only one emulator that runs for the system at all

gizmo98, 2015-01-22:
Hi ProxyCell,

CATSFC is added now. Seems to need 1GHz OC.
https://github.com/petrockblog/RetroPie-Setup/commit/75e636f59bba19dd262f7513649bbd7cc7fa0894

ProxyCell, 2015-01-22:
thanks gizmo98, i already have it compiled and running star fox nice and shiny :)
it currently uses it's own rom folder too, not a huge issue for me, i just copied everything over.

there is a way to alias/sylink the two right?

gizmo98, 2015-01-25:
You could symlink a whole directory.

closed 2015-02-10 by KitchUK