RetroPie/RetroPie-Setup #593: Any mirror site?

SynonZelra, 2015-01-15:
Greetings everyone.

I tried to download retropi via git clone --depth=0 git://github.com/petrockblog/RetroPie-Setup.git

but it's really slow with some timeouts.

So I wonder if there's any mirror site or some other optional location?

Thanks

joolswills, 2015-01-15:
no - perhaps connection issues between you and github ?

try 
```
git clone --depth=1 git://github.com/petrockblog/RetroPie-Setup.git
```

joolswills, 2015-01-15:
the repository is small anyway, so even a complete clone shouldn't take long.

SynonZelra, 2015-01-15:
hi.

and yes, i know it's my connection problem. my isp really sucks. maybe a download is helpful because i can download it via proxy, is it any?

p.s. a ref on my damn isp

C:\Users\Synon Zelra>ping www.github.com -t

Pinging github.com [192.30.252.129] with 32 bytes of data:
Request timed out.
Reply from 192.30.252.129: bytes=32 time=409ms TTL=50
Request timed out.
Reply from 192.30.252.129: bytes=32 time=409ms TTL=50
Reply from 192.30.252.129: bytes=32 time=408ms TTL=50
Reply from 192.30.252.129: bytes=32 time=410ms TTL=50
Request timed out.
Reply from 192.30.252.129: bytes=32 time=409ms TTL=50
Request timed out.
Reply from 192.30.252.129: bytes=32 time=409ms TTL=50
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 192.30.252.129: bytes=32 time=410ms TTL=50
Reply from 192.30.252.129: bytes=32 time=411ms TTL=50

Ping statistics for 192.30.252.129:
    Packets: Sent = 19, Received = 8, Lost = 11 (57% loss),
Approximate round trip times in milli-seconds:
    Minimum = 408ms, Maximum = 411ms, Average = 409ms

joolswills, 2015-01-16:
There isn't a mirror i'm afraid - you could try downloading as a zip archive, but you are going to have similar problems. I'm assuming with that amount of packet loss there is some network problem going on that should be fixed. If not contact your isp.

closed 2015-01-16 by joolswills

njt1982, 2015-01-16:
Jools - do you know how much bandwidth is used per month?

bszafko, 2016-03-19:
I was also having trouble with downloading .img file today. I think github is not that ideal for large files ....

Maybe someone is willing to share the .img file through his google drive account ?

joolswills, 2016-03-19:
@bszafko please ask on the forum - i get very good speed from github - they use amazon AWS so I don't know why you would think they are no good for large files - perhaps a routing/network issue between you and the particular server you got.

bszafko, 2016-03-19:
I got the file finally :) so not a big deal, just letting you know

Now off to play tekken ;)

Thanks!

TechFreak, 2016-04-26:
I also had trouble downloading the disk image (v3.7) from Github. The download keep stalling after 700MB. From what I have read on reddit and here, it seems like a problem for people outside the US. It would be great if you consider distributing also a torrent or any other mirror outside of Github

joolswills, 2016-04-26:
github use amazonaws for their CDN. I am outside of the US and it is very fast here. I would contact your ISP or github perhaps about this though. Not sure I want the additional work of hving to do torrents for every image release too. Alternatively, you can just install from binary on top of an existing raspbian, which will use our server for most of the install.

jeffeld, 2016-10-31:
Adding a +1 for a mirror. I use git all day long for work and no problems. Consistent network error at around 25% for the last 72 hours.

B4rb3rouss, 2017-04-25:
+1 for torrent file.
With DHT, no need for a tracker.

joolswills, 2017-04-25:
We have no plans to provide a torrent ourselves - there is a user on the forum who does it though

https://retropie.org.uk/forum/topic/4400/torrent-download-retropie

joolswills, 2017-04-25:
This issue is closed

SpectreOl, 2020-12-08:
That link to the forum user is great.. not only do they do the torrent, but they keep it up to date too..

i am completely unable to retrieve the image from github, constantly failing the download, but this users torrent fixes that.

unfortunately, the torrent isn't really known about and seeds/peers are few in number. even if you provided a link on the download page, more people would use it and there would be more seeds/peers, which is what torrent really is about..