RetroPie/RetroPie-Setup #597: mupen64rpi needs some changes for retropie

joolswills, 2015-01-19:
We should be installing it in $md_inst - but also currently it builds and installs its own SDL2 in /usr/local/ which is a problem since we build/install SDL via debian package - we don't want it to build against its own SDL2 unless necessary, but if not avoidable it should link statically. I should think linking with our already installed SDL should be possible - maybe we can just not use the build.sh and download the needed plugins ourselves etc.

@gizmo98 I assume we keep this due to better performance than the libretro core or ?

gizmo98, 2015-01-20:
If X11=0 is set SDL2 should not be build and installed. build.sh is broken. I test a fixed one atm.

Test build syntax: 
X11=0 GCC=4.8 ./build.sh

joolswills, 2015-01-20:
I don't think I tried that, but looking at the older code, I'm not sure that works - but check this ticket (towards the end) - a new option has been added which should solve things - please feel free to test/add anything to the issue.

https://github.com/ricrpi/mupen64plus/issues/5

gizmo98, 2015-01-20:
Ok. mupen64plus should find our SDL2 now:
https://github.com/gizmo98/mupen64plus/commit/8dae8228ca10f2f4cf36876ab2f4ae2c740a7c9f

joolswills, 2015-01-22:
I have reworked the module in https://github.com/petrockblog/RetroPie-Setup/commit/6875f32668aaf11e9558dee6c228ba97c784608b

Please give it a test and let me know - will be possible to automatically distribute binaries for it now

In the original post I asked about the performance vs the libretro core - I assume one works better than the other for some stuff or there is a performance difference ? Please can you give some details?

joolswills, 2015-01-22:
btw the module is called mupen64rpi, but the rom folder is n64-mupen64plus - perhaps the module should be renamed to mupen64plus also ? Would be more consistent.

gizmo98, 2015-01-25:
The orginal repo was mupen64plus-rpi. mupen64plus-rpi is deprecated and was superseded by mupen64plus from ricrpi. The file could be renamed.

closed 2015-02-04 by joolswills