RetroPie/RetroPie-Setup #599: Selective Emulators

KitchUK, 2015-01-20:
Would it be possible to have the ability to have ROMs start up in a per game basis? So you can choose what emulator you want to use per game?

joolswills, 2015-01-20:
It isn't currently, but most emulators use different rom folders, so you can selectively put roms in the folder for the emulator you want etc.

joolswills, 2015-01-20:
what you are asking has been discussed briefly before and would be useful, but would require a fair bit of additional work - so it may happen in the future, but there is plan currently to implement it.

KitchUK, 2015-01-22:
Thank you for the response. I had a feeling it might take a lot of work to be able implement this.   Btw, thank you for all your contributions to this project. :)

joolswills, 2015-02-06:
you're welcome :)

closed 2015-02-10 by KitchUK