RetroPie/RetroPie-Setup #600: I don’t have internet access for my Pi, as such I need to add ROMs via USB and a memory pen. Using version 2.4.2, no ROM directories copy over to the USB driver. What am I doing wrong?

petrockblog, 2015-01-20:
From a [user's comment](http://blog.petrockblock.com/download/retropie-sd-card-image-v2-4-small-beta/#comment-1802924900):

> I don’t have internet access for my Pi, as such I need to add ROMs via USB and a memory pen. Using  version 2.4.2, no ROM directories copy over to the USB driver. What am I doing wrong?

I think that we should enable the USB Copy ROM service per-default for the SD-card image.

free5ty1e, 2015-01-20:
Having the USB copy roms service enabled by default would be handy for most situations.  

It would also be nice to have an user-friendly entry in the RetroPie-Setup menu to disable the service as well.  I know we can just go delete the copyroms usb mount script but it feels bad to only have an Enable Copy ROMS service in the menu and not a Disable Copy ROMS service.  

(For instance, I always set up my RetroPie to look for BIOS and ROMS on my USB drive instead of keeping them stored on the SD card, so I can run with a smaller cheaper SD card and not feel bad about copying over huge PSX images)

EatsABurger, 2015-01-24:
I'm running into this problem too - the 2.3 compile had the USB transfer as default - why the change in 2.4.2?  It especially got hairy when I saw all the added emulators and how RetroArch is now under mame-libretro which looks like RetroArch and Mame4All options will both be available for storing instead of just one or the other?

EatsABurger, 2015-01-25:
Playing around some more - is anyone else having trouble reading games (have only tried with MAME4ALL) directly off the USB in 2.4.2?  It's working if I store them on the SD card, but is there a new way of unzipping that won't always be compatible with flash drives?

joolswills, 2015-01-26:
@EatsABurger please open another ticket for that - sounds like a different issue.

@petrockblog I just checked and the copyroms is enabled by default, so there must be another issue. I'll have a test and see.

joolswills, 2015-01-26:
fixed in https://github.com/petrockblog/RetroPie-Setup/commit/e43fbc0190b88e2d2ecadd6ca21a2370273ebab3 - anyone with existing image can do

```
git pull
sudo ./retropie_packages.sh usbromservice
```

to update the script.

jens-b, 2015-02-24:
Sorry for asking here again, or may be i have a suggestion:
I symlink the roms folder from sd card to my usb Harddisk, after that i delete the script 01_retropie_copyroms in the /etc/usbmount/mound.d dir. 
I think this should work? 
After that i am able to copy the roms via smb directly to the usb hard drive and retropie shouldn't copy the roms automatically to the sd card?

May be this could be an option for users who don't want to copy from usb drive to sd card?

closed 2015-03-13 by petrockblog