RetroPie/RetroPie-Setup #602: Simple way for (re-)installing LXDE

petrockblog, 2015-01-21:
We should add a __simple way for installing and configuring LXDE__, if the user want to have a Desktop environment back, e.g., for configuring WiFi - see http://blog.petrockblock.com/download/retropie-sd-card-image-v2-4-small-beta/#comment-1805330041.
This feature request comes up regularly.

joolswills, 2015-01-21:
surely they should then just use raspbian - maybe it;s better we just tell people if you want a desktop - use raspbian ? wifi is pretty simple to configure from console - we could add a module for it, but here are really simple instructions on the raspbian website also.

lxde can be installed by doing "sudo apt-get install lxde" - so already quite simple :)

joolswills, 2015-01-21:
I see they want raspbian settings - I think the answer is to run raspbian then - this is an emulator distro - not a desktop disotro with some emulators.

joolswills, 2015-01-21:
(btw I think you should turn off comments on the download so people use the forum - much easier to help then)

petrockblog, 2015-01-22:
It is ok for me, if we keep LXDE off the official image. However, we should give the users at least the possibility to use the USB copy service then per default (as written in #600), if no network connection is available.

closed 2015-01-22 by petrockblog

joolswills, 2015-01-22:
ill take a look at that when I next get a free moment. Cheers.