RetroPie/RetroPie-Setup #605: tools / configuration menu

joolswills, 2015-01-21:
Currently in emulation if esconfig has been built you get an esconfig menu with esconfig as well as some other configuration scripts made in python.

esconfig needs work as it only handles a couple of emulators.

the menu has others items as mentioned - to switch audio etc.

 I think it would be nice if the menu was called "Tools or Setup or Configuration (you choose)" and we had some additional items in there. The menu would also be installed by default on the image.

My idea is that you would have access to the retropie-setup script itself, but also shortcuts to things like -

configure which emulators use dispmanx
configure retronet settings

id also convert some of the existing python scripts to  bash scripts I think, as a few of them don't really need to be python (they just execute shell commands).

thoughts ? Let me know if I've overlooked anything.

joolswills, 2015-01-23:
as part of this I would also add the raspi-config as an option.

I think we should remove overclocking from the default image btw - just to rule out issues caused by that (and let people overclock themselves)

joolswills, 2015-01-23:
(I have seen reports from people saying it crashes but raspbian works - probably down to a crappy PSU but by removing the overclocking we can avoid the blame and leave it to the user to configure)

petrockblog, 2015-01-24:
> esconfig needs work as it only handles a couple of emulators.

Your are right I think that making the configuration of controllers more user friendly should have one of the top priorities. So many people are struggling with it. However, A core element of ES-Config is the use of AngelScript, which I do not have a clue of (yet).

> I think it would be nice if the menu was called "Tools or Setup or Configuration (you choose)" and we had some additional items in there. The menu would also be installed by default on the image.

I like that idea. In this way we make the various configuration and other supportive utilities more present to the users. And if they do not like them to be shown in the ES directly, they could still remove it from the es_systems.cfg.

> I think we should remove overclocking from the default image btw - just to rule out issues caused by that (and let people overclock themselves)

I was not aware of that until now, but letting the users explicitly decide on their own about the overlocking certainly also makes sense.

PeanutButterBoy, 2015-01-29:
Great suggestions! I think having a setup/configuration/update menu would be very helpful. I am a Linux newbie (as I'm sure many RetroPie users are), and I'm fairly comfortable with the command line, but the idea of having semi-advanced options in the form of a GUI would be extremely useful, and probably cut down on a lot of the "issues" people experience here. Items such as:

1. Setup
 a. Controller Configuration
 b. Emulator Settings (individual for each emulator. Let's user choose amongst advanced options when running an emulator)

2. Update
 a. Update Emulators (automatically check/pull updated emulator binaries)
 b. Themes (simple theme browser/downloader?)

Just a newbie's thoughts.

joolswills, 2015-01-29:
1) is tons of work which is the only reason why it doesnt exist. there are esconfig setting for about 3 emulators.

2) already on the cards. was only possibly recently since the emulator binaries were separated. you can install individual emulators from binary manually though - but it wont automatically update yet.

PeanutButterBoy, 2015-01-30:
1) What I had in mind was any options to append when an emulator is executed. Like for retroarch, you can add "-f" to force fullscreen. So the user could select an option to "Use Fullscreen" when the N64 emulator is run, but may not want that for the NES emulator, for example.

2) I can run the Retropie setup script even though I installed the SD Card Image? I see it's on my device, but I didn't think I could run it without re-installing everything. I guess this is where the newbie kicks in.

joolswills, 2015-01-30:
1) there are already some launch options for emulators - such as switching dispmanx on / off, changing screen resolution, etc, but doing different launch options for every emulator is quite some work. some are configurable in the emulator anyway like vice.

2) yeh the retropie-setup does that already - you can run it on the pre-installed image and update your image.

joolswills, 2015-03-21:
The menu is now in place. 

in regards to some earlier points brought up - we also have some additional control over how emulators are launched, for example controlling the retroarch render resolution. etc. new image out soon.

closed 2015-03-21 by joolswills