RetroPie/RetroPie-Setup #612: Improve startAmigaDisk.sh

BrianGilbert, 2015-01-23:
This allows loading 2 disk games from the menu by selecting the 1st disk it will symling df1.adf as well if the file name contains _disk1.adf or _disk2.adf
```
#!/bin/bash

file="$1"
i=$((${#file}-1))
[[ -z "$file" ]] && exit 1

pushd "/opt/retropie/emulators/uae4all"
rm -fr df*.adf
case $file in
  *_disk1.adf|*_disk1.ADF)
    ln -s $file df0.adf
    ln -s ${file:-5}2${file:$i:4} df1.adf
    ;;
  *_disk2.adf|*_disk2.ADF)
    ln -s $file df1.adf
    ln -s ${file:-5}1${file:$i:4} df0.adf
    ;;
  *)
    ln -s $file df0.adf
    ;;
esac
/opt/retropie/supplementary/runcommand/runcommand.sh 1 ./uae4all "uae4all"
popd
```

BrianGilbert, 2015-01-23:
possibly superceded by https://github.com/petrockblog/RetroPie-Setup/pull/560

joolswills, 2015-02-14:
Going to close this if that's ok - just because the new emulator doesn't support loading from commandline (yet). if/when we add it, it may be enough just to insert the first disk, and others can be done from the gui. 

Thanks.

closed 2015-02-14 by joolswills