RetroPie/RetroPie-Setup #616: Add ES theme as module: ZOID

petrockblog, 2015-01-26:
A user has presented another theme for ES, that he calls "Zoid": http://blog.petrockblock.com/forums/topic/theme-zoid/

I did not have a closer look at that yet, but if the theme looks appropriate and complete, we could think about adding it as another module.

joolswills, 2015-01-28:
is it a new theme or an extension of the simple theme with additional backgrounds ? if its an extension of the simple theme I guess it should be fed upstream + we might want to consider using it as default (if it's good)?

robertybob, 2015-07-14:
Any update on this @joolswills ?

robertybob, 2015-08-11:
FWIW, this needs artwork fro the RetroPie menu. I've knocked up a very rough one if the original author doesn't resurface and want to add his/her own.

joolswills, 2015-08-13:
it is now included. thanks.

joolswills, 2015-08-13:
(In regards to missing retropie artwork etc, please submit a pull request / open a ticket on https://github.com/RetroPie/es-theme-zoid -

closed 2015-08-13 by joolswills

HerbFargus, 2015-08-13:
I added it already- thanks @robertybob  but yes sage advice for future reference