RetroPie/RetroPie-Setup #619: usbromservice and automatic directory creation

joolswills, 2015-01-27:
The current default logic is)

if a usb device is connected that doesn't have a rom folder on it, then sync the rom directories from retropie.

I'm not sure this is ideal since it could created unwanted rom directories without users realising when they plugged in newly formatted usb stick.

I think the default logic would be better as

if a usb device is connected and does have a rom folder, then sync the rom directories from retropie.

Then a user has control over the feature completely. They just need to create a roms folder in the root. 

This seems more logical to me at least. and if a user has a roms folder, they probably do want to pick up any new rom folders from retropie as they are added etc.

free5ty1e, 2015-01-27:
I like the proposed new default logic!

petrockblog, 2015-01-27:
I do not like the idea that I would have to manually create 20+ folders. Or do you mean that the USB device just needs a folder `/roms`at the root of the USB device? That, in turn, would be ok for me - we need to describe that on the wiki pages and the forum, though.

joolswills, 2015-01-27:
just the rom folder :) but it would sync new folders from retropie which the current system doesn't

evilllama, 2015-02-08:
Would this be to sync in both directions? Meaning what is on SD goes to USB and what is on USB goes to SD?

Side "idea", USB based ROM's rather than SD based.
(not a programer or even a novice here, so excuse me for not knowing the correct terms) 
Maybe there could be an option for a pull to the pi to alter the directory the of the ROM's or an image "A" & "B" where as one is stock with usb transfer and one is like mine where as the roms are located @ /meda/usb/(emulator). 
Considering my sd card goes bad often, I don't have to keep re-adding my roms to the sd card, but I do have to alter the es_systems.cfg every time I start over.

Side question: Is petrockblock.com going to be back up soon?

joolswills, 2015-02-08:
it would sync new rom folders, but not any files.

you can symlink/mount the roms folder to wherever you want... you can have it on a usb disk or even over the network.

website should be back up soon I believe. It is being worked on currently.

joolswills, 2015-03-21:
This is done to the point that rom folders will always be synced if the usb drive has a "retropie" folder. Closing as the main functionality of the original issue is implemented.

closed 2015-03-21 by joolswills