RetroPie/RetroPie-Setup #623: retroarch is failing to build the first time, works the second time

ProxyCell, 2015-01-28:
I have uploaded a truncated log file here: http://www.iruberleet.org/files/run_21012015_1426_retroarchFAIL.log

I am working off of a fresh raspbian image, everything up to date, bla bla bla.
I ran this about a week ago but I don't believe the changes since then would have touched upon this, please correctly me if I am wrong.

When I first tried to compile it, I compiled it from the source build menu and I had selected many other options. The only things to come before this would have been FBA (not libretro) and the GBA emulator GPSP. Retroarch fails (please see the log file above) and continues on to compiling other things. Everything else works perfectly and then I just re-run the source build for only retrarch and everything is good.

I have been testing this setup for a while now and couldn't find any problems with it so it appears to be safe to do what I just did. If this can be verified by someone else that would be great, please use a clean raspbian image from the official sources. I'm not sure what the cause of this is but I may try to skip compiling it with everything else and simply do it after everything else is done to see if that changes anything for me.

gizmo98, 2015-01-31:
It cannot build sdl_gfx.c? Could we working if libsdl1.2-dev or SDL2 is added as a dependency.

joolswills, 2015-02-04:
closing as resolved. thanks all.

closed 2015-02-04 by joolswills