RetroPie/RetroPie-Setup #624: Background sound emulationstation

expandables2, 2015-01-28:
Hi is there a way i can add background sound to emulationstation version 1? I see it has scroll sound and select sound is there a way i can edit the code to have background sound? I am using retropie version 1

expandables2, 2015-01-30:
Any reply please

ProxyCell, 2015-01-30:
hi expandables2 - i'm not sure really, this isn't the Emulation Station repository :)
please head on over to: https://github.com/Aloshi/EmulationStation

joolswills, 2015-02-01:
I don't think it supports background sound, and the 1.x version isn't developed I think. for emulationstation 2.x I would ask over at their forum as mentioned. thanks.

closed 2015-02-01 by joolswills

expandables2, 2015-02-01:
I am asking how can i be able to develop version 1 of emulation station to have background sound. Since it is open source it should be able to am guessing.