RetroPie/RetroPie-Setup #625: Scraping NES roms breaks NES emulator

PeanutButterBoy, 2015-01-29:
* *Crossposting from EmulationStation*

I recently put together a Retropie (Raspberry Pi w/ Retropie SD card image). I loaded a few SNES and NES roms and used the built-in scraper to pull data for all of the roms; it pulled all the data fine. The problem is that none of the 20+ NES roms will load - I get a "failed to load rom" error every time. The SNES roms load perfectly fine. So here's what I did:

Deleted "gamelist.xml" in the "nes" folder and replaced it with a new (blank) "gamelist.xml"

Deleted all images that the scraper pulled (I'm not sure this was necessary after deleting gamelist.xml).

After doing this, all of the NES roms work fine! The problem is they don't have the scraped data, which was a nice addition. I searched but couldn't find this issue anywhere. Is this a known bug? Has anyone else experienced this?

joolswills, 2015-03-21:
Unknown - but it might be worth posting on the forum regarding this. 

Is this still an issue with the latest image btw ?

joolswills, 2015-07-13:
closing due to no feedback.

closed 2015-07-13 by joolswills