RetroPie/RetroPie-Setup #628: Ensure compatibility with Raspberry Pi 2

robertybob, 2015-02-03:
I read today that the current RetroPie images dont work with the new Pi. Is it a big job to begin to get it compatible? Do we know if @aloshi 's Emulationstation will work as well?

Aloshi, 2015-02-03:
I ordered an RPi 2 this morning, so I intend to support ES on it. Although it probably won't get here for weeks since it's super back-ordered.

joolswills, 2015-02-03:
the image just needs a minor update to pull in the armv7 kernel and it should work on both. Will do this when I get a free moment. thanks. if you can;t wait, you can always just install retropie-setup on top of an already updated raspbian.

davcri, 2015-02-03:
Do you have any ideas on how to improve retropie for the new RPi 2? I'm thinking about new emulators like PPSSPP. This should work quite well on the new ARMv7 quad-core, right ?

joolswills, 2015-02-03:
We could add the ability to optimize for the new cpu if building from source. I doubt any emulators will be able to utilise the multiple cores of the newer cpu - they are generally single threaded. So I guess the speedup will be around 1.5x or so (which is thefigure given for single core performance improvement). I can't tell you how well a specific emulator will run.

davcri, 2015-02-03:
I didn't know that all emulators were single-threaded. What a shame :cry: Thanks for the info.  

PPSSPP has an experimental multi-threaded option, but it seems that it isn't working well at the moment.

robertybob, 2015-02-04:
Judging by the comments in the forum, it seems to work fine on the Pi. Might as well close this now :)

closed 2015-02-04 by robertybob