RetroPie/RetroPie-Setup #633: Raspi Version 2b

JCRodger, 2015-02-07:
Hi,

how would i go about compiling for the new pi b2? would like to compile the emulators etc
to use armv7.

C-Fu, 2015-02-07:
https://drive.google.com/file/d/0B_knGioK16E6c0d4MDZCRUlqclU/view

AFAIK this is the beta compiled image. seems to be working for me at least.

Adri Idzwan Mansor
General Manager
Iron Ore Division
Malaysian Coal & Minerals Corporation Sdn Bhd

joolswills, 2015-02-07:
to build for pi2 - which is rather new untested so far you can experiment with
```
sudo __platform=rpi2 ./retropie_setup.sh
```

some emulators may fail to compile currently, as I'm still working on support.

C-Fu, 2015-02-10:
When I tried that, some text came up, and it said something about compiling
(I think?) using gcc 4.7?

I don't really know what I'm talking about, but I suppose it would be
better to update the script to compile using gcc 4.8 as it supports armv7
or NEON or both? and if possible, how do I force the script to use gcc-4.8?

I read somewhere in RPi forum that people are "experimenting" with
gcc-4.8 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard

but hey, what do I know, just trying to learn :)

Thanks for the good work btw!

joolswills, 2015-02-10:
gcc 4.7 supports armv7 + neon - else it would complain regarding the parameters. we are using exactly the parameters you mention above - just not with gcc 4.8

Once I have my rpi2 board I will be able to test the code I am producing - currently I am working blind (apart from being able to test the binaries partly work in an emulated environment)

C-Fu, 2015-02-11:
I see... thanks for the clarification. I dived right into compiling
manually (for the first time :D) using gcc 4.8, and all seems to be
compiling just fine (except the Mame emulator I think).

After forcing retropie_setup.sh to compile using gcc 4.8, I tried
Setup>ES-Config and while looking at the compiling logs (?) I noticed few
odd things

/tmp/ccugGWBw.s:3124: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccugGWBw.s:3161: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccugGWBw.s:3318: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccugGWBw.s:3356: Warning: swp{b} use is deprecated for ARMv6 and ARMv7

/tmp/ccugGWBw.s:7802: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccugGWBw.s:7839: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccugGWBw.s:8071: Warning: swp{b} use is deprecated for ARMv6 and ARMv7
/tmp/ccugGWBw.s:8109: Warning: swp{b} use is deprecated for ARMv6 and ARMv7

Not sure if it's any help though. Thanks for the awesome work!

closed 2015-02-14 by joolswills