RetroPie/RetroPie-Setup #637: RetroPie v 2.5 - Xin Mo Controller issues

daveireland20, 2015-02-10:
Hi guys 

Had a small issue with the Xin Mo dual arcade usb controller since updating retro pie to 2.5 and using it on a RPi2. For some reason during configuration in retroarch config and even during use in emulation station and the various emulators one movement of the joystick registers as multiple. Sort of hard to describe however my best example would be during the retro arch controller configuration you press the appropriate button as displayed in the screen and then the next key to be configured will appear. Tilting the joystick in any direction to register it effectively speeds through the other key configurations , almost as if it is registering multiple times. 

Any ideas as to what may be going on ? And a potential remedy if so ? 

Cheers

clownfish360, 2015-02-15:
I am having the same issue.  When I move my joystick it move the emulationstation menu several items at a time.  I am not able to move one at a time unless I use the keyboard.

joolswills, 2015-02-15:
it is an upstream kernel issue - nothing we can do about it really until it is fixed.

https://github.com/raspberrypi/linux/pull/828

daveireland20, 2015-02-18:
There is a fix for this issue using a custom kernel build created by feederchain over on the RetroPie forum.  http://blog.petrockblock.com/forums/topic/autofire-on-the-axis/#post-87676

closed 2015-02-18 by daveireland20