RetroPie/RetroPie-Setup #639: mupen64plus fixes for pi2

joolswills, 2015-02-10:
@gizmo98 

Currently it doesn't work when built for the pi2

Still going through it but opening this ticket so you have all the info as you have put much of the work into this.

first things I notice are
mupen64plus/mupen64plus-video-gles2n64/projects/unix/Makefile
mupen64plus/mupen64plus-video-gles2rice/projects/unix/Makefile

both look for armv6 to decide whether to add pi include/libraries - would be much better if all this "intelligence" was removed from the makefile and it just took a build parameter for platform. Especially for the case if someone ever wanted to cross compile.

Would you mind handling this / reporting it upstream etc - and in the meantime, I will rewrite the makefiles as needed from the module to try and get it all working. Cheers!

joolswills, 2015-02-11:
the above commit fixes mupen64plus video out for the pi2. So it works now, however on my dev set up (hdmi to dvi to 4:3 monitor @1280x1024), it outputs to a small area at the bottom left. There are a whole bunch of configs for gles2n64.conf I see, so I'll have a play - but happy to receive any advice.

joolswills, 2015-02-11:
ok. my bad. all is good - I just needed to set the correct screenmode.

gizmo98, 2015-02-11:
I will report it upstream. If there is no response we can use my repo.

gizmo98, 2015-02-14:
Ok. There are four PRs now:
https://github.com/ricrpi/mupen64plus-video-gles2rice/pull/4
https://github.com/ricrpi/mupen64plus-video-gles2n64/pull/6
https://github.com/ricrpi/mupen64plus-core/pull/3
https://github.com/ricrpi/mupen64plus-audio-omx/pull/1

If everything is merged the mupen64plus module must  be modified.

joolswills, 2015-02-14:
ok looks good. thanks for your work.

gizmo98, 2015-02-15:
One more info. Video plugins do not respect screen resolution atm. This can be fixed but there are two solutions. Modify SDL2 so we can use dispmanx upscaling or modify the video plugin so render resolution = screen resolution (slow).
https://github.com/ricrpi/mupen64plus-video-gles2n64/issues/7

gizmo98, 2015-02-16:
All PRs are merged!

joolswills, 2015-02-16:
Just a heads up - I was wondering why my build of mupen64-libretro wasn't working since you had been working on it on your repo. I see you have a fixed branch (patch-2) which is default. However due to the build system and chroot problems with git the script utilises downloading tar in such environments, and it will default to use the master branch - so it's not enough to set the default on github, you need to provide the branch in the retropie-setup module. It is also more verbose to add it, as you can see from the script precisely which is used and there will be no confusion if a default branch is changed.

PS. please can you start using somewhat descriptive branch names for future pull requests. patch-1 patch-2 provides little information, and makes the merge commits ugly.

gizmo98, 2015-02-16:
Sorry Jools! I will do my best next time. 

I renamed patch-2 to rpi2_fix. But as you said it is necessary to provide the branch in mupen64plus-libretro module.

joolswills, 2015-02-16:
I will add it. thanks.

joolswills, 2015-02-16:
ok done and built. I'm getting  better performance on my pi2 with the libretro core than the default mupen64plus

joolswills, 2015-02-16:
I mentioned before I think that there is no video output with the rice plugin and star wars racer (and possibly other games). might it be worth using the other output driver by default as it is faster and doesn't have this problem?

joolswills, 2015-02-16:
aah sorry must have been some configuration / build  issue, it's using the other plugin by default now.

gizmo98, 2015-02-17:
mupen64plus --> default plugin gles2n64 (because it is faster)
mupen64plus-libretro --> default plugin rice (because Zelda OoT does not crash with rice. gles2n64 should be faster.
mupen64plus-testing --> default plugin gles2n64 (because it is faster and has a fullscreen option now)

Rice is the only plugin running Zelda OoT. I think about rom folders for every video plugin because gles2n64 looks better and is faster but some games are not working. On the other hand rice is much slower but is more compatible with some games. 

You could also check out mupen64plus-testing. If you change resolution with runcommand.sh "m" you can even set 1080p without changing any ini setting.

joolswills, 2015-03-21:
This can be closed now. Thanks for all the infos.

closed 2015-03-21 by joolswills