RetroPie/RetroPie-Setup #644: Add an image for NOOBS!

belvsdragon, 2015-02-12:
I don't understand why there is no image for NOOBS. The Lakka Project made an image for BerryBoot and NOOBS. I would be happy to see the same for RetroPie.

https://github.com/libretro/Lakka/wiki/Alternative-installation-methods
https://github.com/raspberrypi/noobs#how-to-create-a-custom-os-version

joolswills, 2015-02-14:
There is no image for NOOBS because no-one has made one yet.

C-Fu, 2015-02-14:
Is there a way to make one? I only have access to one linux machine (RPi &
RPi 2) I remembered last time the only way was to use other linux machine
to create a sqashfs image file that NOOBS (or was it berryboot?) can read.

Adri Idzwan Mansor
General Manager
Iron Ore Division
Malaysian Coal & Minerals Corporation Sdn Bhd

joolswills, 2015-02-14:
I will look into automating the creation of a compatible image (when I have time). thanks.

joolswills, 2015-10-10:
We now have berryboot images - I believe this should be sufficient. No plans to maintain any other images for now.

closed 2015-10-10 by joolswills