RetroPie/RetroPie-Setup #654: After pairing one PS3 controller, the second cannot be paired

kasgore, 2015-02-15:
We were able to identify several bugs and find fixes for them in the following forum conversation.
http://blog.petrockblock.com/forums/topic/multiple-ps3-controller-via-bluetooth-frustration/#post-87517

List of bugs:
1. Cannot pair a second PS3 controller after pairing the first one
2. Bluetooth adapter is in state down after fresh image copy to SD card from download of 2.5
3. Not enough time to pair bluetooth controller before emulationstation boots and you get the "no controllers detected" which prevents keyboard free operation and convenience. 

Fixes

Issue 1
Before pairing the first controller if you run "hciconfig" the mode is "Up Running pscan:. After pairing the first controller the bluetooth adapter exits discovery mode and hciconfig then only reads "up running"

Fix 1
If you change the mode to PISCAN by running "sudo hciconfig hci0 piscan" after pairing the first controller the bluetooth stays in the discovery mode and it works as expected for the pairing of the second controller.
I fixed this by inserting "hciconfig hci piscan" into the rc.local file just before the exit 0 line.

Issue 2
After a fresh install to the SD card and the first boot the bluetooth adapter is in the state down if you run "hciconfig" this could be confusing to new users and prevents a trouble and hassle free install if the intent is to use a bluetooth controller

Fix 2
If you run the command "sudo hciconfig hci0 up" this issue is simply fixed so I am guessing before the image was imaged the bluetooth adapter simply needs to be told to be up so when users install the image it will continue to be that way.

Issue 3
Especially with the RPI2 there is simply not enough time to pair the controllers before the "no controllers connected" dialogue pops up. I propose there is a simple option to enable or disable the popup or another workaround is done, this prevents the user from being able to sit back on the couch with wireless controllers and no keyboard on the PI to hit F4 and pair and then run the emulationstation comand

Fix 3
A simple workaround we found was to setup the keyboard as a controller and then even after the keyboard is removed the popup never comes up again. I am not sure this is the best fix but it does work.

free5ty1e, 2015-02-17:
I can confirm that these workarounds are functional.  Once "sixpair" has set both of my PS3 controllers' BT master address to the Pi's BT adapter, I can boot up the system with no controllers attached, wait for Emulationstation to start up, and (issue 3 workaround also applied) at this point turn on both controllers and watch them pair correctly to js0 and js1.  I can then control Emulationstation with either controller or the keyboard.  This is excellent!

gizmo98, 2015-02-18:
Please modify the PS3 controller module and do a PR. You can find the module here:
https://github.com/petrockblog/RetroPie-Setup/blob/master/scriptmodules/supplementary/ps3controller.sh

As i understand Fix1 should be added in line 44 and Fix2 in line 33?

There is another way to pair a PS3 controller. I use the bluez with sixaxis support instead of sixpair atm. http://blog.petrockblock.com/forums/topic/using-the-latest-bluez-for-ps3-bluetooth-how-to/

free5ty1e, 2015-02-20:
Ooh, I'm gonna have to try out bluez with sixaxis support!  I was not aware of such a capability without sixad.  Thanks!!
Once I have something that seems to work reliably for me every time I try on a fresh system and then repeatedly from the couch thereafter, I'll most def. put together a PR to assist the PS3 goodness!

PeterPablo, 2015-02-20:
subscribing

NaWer, 2015-03-15:
In the PS3 install script, maybe we can add in `/etc/udev/rules.d/99-sixpair.rules` something like that :
`DRIVER=="usb", SUBSYSTEM=="usb", ATTR{idVendor}=="054c", ATTR{idProduct}=="0268", RUN+="/opt/retropie/supplementary/ps3controller/sixpair"`
And of course `hciconfig hci0 piscan` in `/etc/rc.local`

free5ty1e, 2015-03-17:
@NaWer awesome!  I must try this out.  

The only downside to this that I can see is it would pretty much completely disable using a PS3 controller over USB.  

(As soon as `sixpair` is executed, the controller no longer responds over USB until the cable is removed and reconnected.  

With this rule in place, this would ensure that the controller is always immediately disconnected from USB as soon as it is recognized, so it would ONLY function over bluetooth.  

This is probably fine, but I put this out there in case someone has a solution to `sixpair` killing the USB controller input.)

m42i, 2015-03-27:
Just wanted to add short comment.

It`s enough to use pscan instead of piscan. The i makes the RPi discoverable which might not be what you want.

gizmo98, 2015-04-04:
Could someone test my "optimized" PS3 controller module?
https://github.com/gizmo98/RetroPie-Setup/commit/9798c0220242e146ce8aa78b1eb6547a5e3f73b4

m42i, 2015-04-04:
@gizmo98 I tested your update on my test installation. I didn't do a clean install but uninstalled all bluez5, bluez and sixad stuff.

I had to change two lines:
64: Removed ! from the condition, also my hciconfig is in /usr/sbin
99: Added a ! to the condition.

Everything looks fine, pscan on bt-dongle, bluetooth service running, but I didn't get the controllers paired at first.

/var/log/daemon.log showed a lot of this, when I press the PS button.
    
    retropie bluetoothd[3315]: Refusing input device connect: No such file or directory (2)

And I didn't see anything under /var/lib/bluetooth/*/ either.

After a reboot it works, though I noticed the bluetooth service wasn't starting automatically. Also I still only get two controllers paired at a time. Interestingly every controller I pair after the first one lids the second LED. After I disconnected a previous one.

gizmo98, 2015-04-04:
Such a shame. I removed the senseless "!". It seems to run on a clean fresh installation. Plug in your PS3 controller and the connection is setup!

joolswills, 2015-10-10:
Please could someone test pairing two controllers with the 3.1 image. thanks.

gizmo98, 2015-10-10:
In my opinion there is still a problem. If a ps3 controller is connected "hciconfig hci0 pscan" will be disabled. The bluetooth dongle must be discoverable. Raspbian seems to have a problem: http://raspberrypi.stackexchange.com/questions/32451/usb-bluetooth-dongle-loses-its-pscan-state-when-connected-via-bluetooth-serial

So if a controller is connected we must enable pscan again. We could add a udev rule (if bluetooth controller is connected enable pscan again) or add a cron job which checks or enables pscan periodically.

joolswills, 2015-10-10:
I thought we did enable pscan again - hence you rpoint about that line which was removed (and re-addded) ?

gizmo98, 2015-10-10:
As i think of it's just a workaround to enable pscan again if you connect a ps3 controller over usb. 
1. We start pscan if a bluetooth dongle is present with udev rule->bluetooth.sh
2. We configure a ps3 controller with udev rule->ps3pair.sh. sixpair writes the mac address of your bluetooth dongle to the ps3 controller. An discoverable (pscan) bluetooth dongle is necessary for successful (first time) configuration. If for "unknown" reason pscan is disabled we added "hciconfig hci0 pscan".
3. After configuration "pscan" enabled is necessary to pair a controller (press PS button).
4. If a ps3 controller is paired pscan gets disabled. 
5. Care must be taken pscan is always enabled (check with cron or udev rule) to ease use of a second ps3 controller.

Enable pscan in ps3pair.sh is a workaround to get a second controller working if you connect it to your usb port (udev rule will be triggered). Everytime you reboot your pi you must connect a second controller over usb once.

joolswills, 2015-10-10:
I see - Will look into possible solutions. Maybe I can get hold of a second controller for some testing.

gizmo98, 2015-10-13:
Other scenario. Same problem. Disable ps3 controller (press 10s ps button) and try to reconnect. Does not work because bt dongle is not discoverable anymore.

gizmo98, 2015-10-18:
Could be a workaround.
1. add script /opt/retropie/supplementary/ps3helper.sh
`#!/bin/bash
if hciconfig dev | grep -q "hci0"; then
    if hciconfig | grep -q -v "pscan"; then 
        hciconfig hci0 pscan
    fi
fi`
2. add udev rule 99-ps3vinput.sh:  
`# Enable pscan again
SUBSYSTEM=="input", ATTR{name}=="PLAYSTATION(R)3 Controller", RUN+="/opt/retropie/supplementary/ps3controller/ps3helper.sh"`

gizmo98, 2015-10-20:
Should be ok now.

closed 2015-10-20 by gizmo98

eddywebs, 2016-05-12:
I believe this is still an issue with retropie 3.7 using gasia or clone ps3 controller ? 
It actually crashed the entire system:
Steps to reproduce:

 - Pair 1 ps3 controller to the retropie (gasia/clone)
 - Pair 2nd controller by connecting it to the raspberrypi using usb cable, disconnect usb cable press the PS3/home button the lights on 2nd controller keeps on flashing and eventually the system crashess.

eddywebs, 2016-05-15:
Since nobody showed interested I tried working on it in, and confirm there there no more crashes after updating to linux kernel to 4.4 >> https://github.com/raspberrypi/linux/issues/1360#issuecomment-217863895

However any additional controller still fail to pair on retropie, this time there is no bluetooth crash its just no paring as 2nd or 3d controller.

gizmo98, 2016-05-15:
No dev has a gasia controller. As far as i know the "gasia only" installation option gives multiple gasia controller support. The "clone support gasia" installation option gives you original and gasia controller support.

eddywebs, 2016-05-15:
@gizmo98 Thanks for the tip will try the gasia only .

coproc79, 2017-01-12:
I have search and have not found any information on how to reassign the PS3 controller to a different number.
I have a fresh install, and a keyboard currently connected.
When ES loads and I power own the controller it defaults to the 2nd controller slot. 
Within ES that isn't an issue, but playing any of the roms, the emulator doesn't register any controller.
I don't have another PS3 controller to pair either, and on a side note, connecting the USB works just fine.

Anyone know of a way to change the assignments?

joolswills, 2017-01-12:
@coproc79 this is not the right place for support - please use the retropie forum - https://retropie.org.uk/forum/

mike12341234, 2017-03-24:
hello

mike12341234, 2017-03-24:
(ps3 shaman)I'm having an issue connecting a second bluetooth remote first one connects fine when i connect second one it connects and can only be used for 5 seconds before both of them fail the lights on remotes stay on but nothing happens ....when i connect first one it works flawlessly...i had another dongle which broke installed previously that ran 4 blue tooth connections no problems....is it the dongle or script

reply

thiagoufg, 2017-07-23:
On retropie 4.2, my ps3 controller would not reconnect if it disconnected after some time.
Neither would it allow me to connect a second controller.

I fixed it with the following udev steps...

Create the file ps3reconnect in your home folder with this:

sudo /usr/bin/sixad --stop
sudo killall sixad &
sudo service bluetooth restart
sudo nohup /usr/bin/sixad -s &
sudo echo $(date) > oi.txt

Make it executable:

chmod +x ps3reconnect

Add the following to the last line of /etc/udev/rules.d/99-sixad.rules:

SUBSYSTEM=="bluetooth",ACTION=="remove",RUN+="/home/pi/ps3reconnect"

Then run the following to reload udev rules:

udevadm control --reload-rules && udevadm trigger

By the way, I'm running retropie 4.2 on a raspberry pi 3, with a fake ps3 controller (gasia). The gasia controller only connected after i followd [this tutorial](https://raspberryblog.de/?p=1870&page=2).

Be careful that your modifications to both files use "Linux style line endings" (which is LF only), not Windows style (whi is CR and LF). Otherwise, your scripts won't work.