RetroPie/RetroPie-Setup #655: Kodi PVR addon & Joystick issues with RetroPie

C-Fu, 2015-02-15:
So after accidentally discovering a Kodi installer (?) within RetroPie-Setup (how awesome! Really appreciate it since it took me a few days just trying my luck installing XBMC and Kodi on my own), I found out that the PVR/LiveTV function was disabled, or unavailable. I believe it's because the "installer" did not install the necessary package for it to function/enable, like Simple IPTV or TvHeadend. So I did

sudo apt-cache search kodi-pvr*

and it returned 

kodi-pvr-argustv - XBMC PVR Addon Argustv - PVR API:1.9.2
kodi-pvr-demo - XBMC PVR Demo Addon - PVR API:1.9.2
kodi-pvr-dvbviewer - XBMC PVR Addon Dvbviewer - PVR API:1.9.2
kodi-pvr-iptvsimple - XBMC PVR Iptv Addon - PVR API:1.9.2
kodi-pvr-mediaportal-tvserver - XBMC PVR Addon Mediaportal Tvserver - PVR API:1.9.2
kodi-pvr-mythtv - XBMC PVR Addon MythTv - PVR API:1.9.2
kodi-pvr-nextpvr - XBMC PVR Addon NextPvr - PVR API:1.9.2
kodi-pvr-njoy - XBMC PVR Addon Njoy - PVR API:1.9.2
kodi-pvr-tvheadend-hts - XBMC PVR Addon TvHeadend Hts - PVR API:1.9.2
kodi-pvr-vdr-vnsi - XBMC PVR Addon VDR Vnsi - PVR API:1.9.2
kodi-pvr-vuplus - XBMC PVR Addon Vuplus - PVR API:1.9.2

so I typed

sudo apt-get install kodi-pvr-iptvsimple

and voila, PVR/LiveTV function is available again!

2nd issue is with joystick support. While Retropie & EmulationStation found my joystick, Kodi doesn't. As I understand, there is no support for joystick in Kodi. But then I found this:

https://github.com/garbear/peripheral.joystick

Apparently Kodi devs are working for a new type of addon for Peripherals. So I tried compiling it as per the instructions, but doesn't work.

Reason I asked is that it would be awesome for me (and I'm sure for a few others as well) to have RetroPie for the gaming backend, EmulationStation 2 as the frontend, Raspbian for Desktop, and Kodi for a full live streaming and media consumption device.

Sebubu, 2015-07-19:
+1

Any news?

necrophcodr, 2015-10-05:
Kodi (at least as of 15) supports joystick and gamepads for controlling the interface. Tested on a Void Linux setup with Kodi 15.1 installed, it seems to work, but required me to install certain XML files into the `~/.kodi/userdata/keymaps/` directory. After doing so, it worked perfectly. However, installing these files in the corresponding directory on RetroPie, even with the same version of Kodi, does not work, suggesting that either the build of Kodi used is compiled without joystick/gamepad support, or has downright disabled the support for these.

To enable support for this in Kodi, it needs to be `./configure`'d with `--enable-sdl` and `--enable-joystick`.

HerbFargus, 2015-10-05:
Submitted a request here:

http://michael.gorven.za.net/raspberrypi/xbmc#comment-1907

necrophcodr, 2015-10-05:
@HerbFargus sounds good, gonna follow that thread as much as I can for updates, so I can pull it from the unstable repo, or where ever it lands.

pancakeykakes, 2015-10-05:
I just use xboxdrv with arguments to bind controller buttons/axis to keyboard commands when Kodi is launched, and xboxdrv with basic arguments for emulators to pick it up when in RetroPie.

I have Kodi and RetroPie installed as standalone environments on top of Wheezy.

necrophcodr, 2015-10-06:
@pancakeykakes does it work with controllers other than the Xbox controllers?

pancakeykakes, 2015-10-06:
I have not tried with other controllers. I mean, its called xboxdrv so I wouldn't put too many eggs in that basket.

FishTest, 2016-03-08:
I have the same problem...

joolswills, 2016-04-09:
Repo with Kodi 16 added now for RPI - which has patches for joypad support. Requires a manual configuration currently though.

closed 2016-04-09 by joolswills

necrophcodr, 2016-04-11:
What kind of manual configuration does it require? No need to go into full detail about it, but some hints, or perhaps links, might be ideal, especially for people following this issue.

HerbFargus, 2016-04-11:
On the wiki. If you need further help ask o. The new forum.

https://retropie.org.uk/forum/