RetroPie/RetroPie-Setup #659: Having problem on "First Installation" ( libsdl1.2-dev)

ArdaOzturk, 2015-02-16:
Hi,

I'm very new to this project and this weekend I tried to install it for the first time. While trying to install in binary mode or full installation I get this error;

The following packages have unmet dependencies:
 libsdl1.2-dev : Depends: libsdl1.2debian (= 1.2.15-5) but 1.2.15-8rpi
is to be installed
E: Unable to correct problems, you have held broken packages.
Could not install package(s): libsdl1.2-dev. Aborting now.

I've tried several things such manually install related libs but I could not succeed. You can find whole ouput below;

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for AdvanceMAME
= = = = = = = = = = = = = = = = = = = = =

Did not find needed package(s): libsdl1.2-dev. I am trying to install them now.

= = = = = = = = = = = = = = = = = = = = =
Installing SDL 1.2.15 with rpi fixes and dispmanx
= = = = = = = = = = = = = = = = = = = = =

--2015-02-14 20:40:35--
http://downloads.petrockblock.com/retropiebinaries/rpi1/libsdl1.2debian_1.2.15-8rpi_armhf.deb
Resolving downloads.petrockblock.com (downloads.petrockblock.com)...
46.252.21.198
Connecting to downloads.petrockblock.com
(downloads.petrockblock.com)|46.252.21.198|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 203928 (199K) [application/octet-stream]
Saving to: `libsdl1.2debian_1.2.15-8rpi_armhf.deb'

     0K .......... .......... .......... .......... .......... 25%  349K 0s
    50K .......... .......... .......... .......... .......... 50%  898K 0s
   100K .......... .......... .......... .......... .......... 75% 2.18M 0s
   150K .......... .......... .......... .......... ......... 100% 16.1M=0.2s

2015-02-14 20:40:40 (887 KB/s) -
`libsdl1.2debian_1.2.15-8rpi_armhf.deb' saved [203928/203928]

--2015-02-14 20:40:40--
http://downloads.petrockblock.com/retropiebinaries/rpi1/libsdl1.2-dev_1.2.15-8rpi_armhf.deb
Resolving downloads.petrockblock.com (downloads.petrockblock.com)...
failed: No address associated with hostname.
wget: unable to resolve host address `downloads.petrockblock.com'
(Reading database ... 78036 files and directories currently installed.)
Preparing to replace libsdl1.2debian:armhf 1.2.15-8rpi (using
libsdl1.2debian_1.2.15-8rpi_armhf.deb) ...
Unpacking replacement libsdl1.2debian:armhf ...
dpkg: error processing libsdl1.2-dev_1.2.15-8rpi_armhf.deb (--install):
 cannot access archive: No such file or directory
Setting up libsdl1.2debian:armhf (1.2.15-8rpi) ...
Errors were encountered while processing:
 libsdl1.2-dev_1.2.15-8rpi_armhf.deb
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Hit http://mirrordirector.raspbian.org wheezy Release.gpg
Hit http://raspberrypi.collabora.com wheezy Release.gpg
Hit http://mirrordirector.raspbian.org wheezy Release
Hit http://raspberrypi.collabora.com wheezy Release
Hit http://mirrordirector.raspbian.org wheezy/main armhf Packages
Hit http://raspberrypi.collabora.com wheezy/rpi armhf Packages
Hit http://archive.raspberrypi.org wheezy Release.gpg
Hit http://mirrordirector.raspbian.org wheezy/contrib armhf Packages
Hit http://mirrordirector.raspbian.org wheezy/non-free armhf Packages
Hit http://archive.raspberrypi.org wheezy Release
Hit http://mirrordirector.raspbian.org wheezy/rpi armhf Packages
Hit http://archive.raspberrypi.org wheezy/main armhf Packages
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en_US
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en
Ign http://archive.raspberrypi.org wheezy/main Translation-en_US
Ign http://archive.raspberrypi.org wheezy/main Translation-en
Ign http://mirrordirector.raspbian.org wheezy/contrib Translation-en_US
Ign http://mirrordirector.raspbian.org wheezy/contrib Translation-en
Ign http://mirrordirector.raspbian.org wheezy/main Translation-en_US
Ign http://mirrordirector.raspbian.org wheezy/main Translation-en
Ign http://mirrordirector.raspbian.org wheezy/non-free Translation-en_US
Ign http://mirrordirector.raspbian.org wheezy/non-free Translation-en
Ign http://mirrordirector.raspbian.org wheezy/rpi Translation-en_US
Ign http://mirrordirector.raspbian.org wheezy/rpi Translation-en
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libsdl1.2-dev : Depends: libsdl1.2debian (= 1.2.15-5) but 1.2.15-8rpi
is to be installed
E: Unable to correct problems, you have held broken packages.
Could not install package(s): libsdl1.2-dev. Aborting now.

joolswills, 2015-02-16:
nothing wrong. it didn't have all the dependencies, so corrected it right after

joolswills, 2015-02-16:
Sorry, should have put my glasses on disregard my first comment. 

you had a network issue during installation, and it failed to download a package. are you having isp problems ? you can see the dns failed/ it isn't a problem of the script, but a networking/dns issue your end.

joolswills, 2015-02-16:
maybe change the dns on your router to 8.8.8.8 / 8.8.4.4 if your isp ones are unreliable.

ArdaOzturk, 2015-02-16:
I had no network issues as far as I see. Also I've tried to update and retried. Nothing changed. But today I updated the script and tried again. There are stillbsome messages like error but installation seems to be continuing.

joolswills, 2015-02-16:
there was a dns lookup failure. whether your end, your isp, some routing issue or something else. impossible to know the cause, but there is unfortunately nothing that can be done about that from our end (so long as we make sure the retropie server is working etc). thanks.

if you are getting intermittent dns its often a good idea to use some different dns servers. often ISP provided dns servers are unreliable.

closed 2015-02-18 by joolswills