RetroPie/RetroPie-Setup #662: FUSE questions

space-invader, 2015-02-16:
Hi.

I want to know if it is posible to start the spectrum games directly from the front end, what kind of file must be used and if it is posible to make a keyboard mapping as in DOSBox.

Sorry for my bad English.

Thanks in advance!

joolswills, 2015-02-18:
this should be fixed. if you update your retropie-setup script and then run

```
sudo ./retropie_packages fuse configure
```

it should fix up the emulationstation config

joolswills, 2015-02-18:
in regards to mapping you need to check the fuse documentation. I will be updating the retropie version of fuse soon.

space-invader, 2015-02-18:
OK, I will try it.

Thanks!

joolswills, 2015-02-18:
the fuse emulator has just been updated, so if you use option 5 in retropie-setup menu, you can install the latest version.

closed 2015-02-18 by joolswills