RetroPie/RetroPie-Setup #663: RPI2: Build using all available cores

sreichholf, 2015-02-17:
Currently on the rpi 2 only 1 core is used to compile the sources.
I guess it would be a nice speedup on the RPI2 when compiling with

make -j`nproc` ....

This will launch "number-of-cores"-instances of make to speedup compiling.

joolswills, 2015-02-17:
this is already supported but must be enabled manually. pass MAKEFLAGS="-jX"  to the script. 

```
sudo MAKEFLAGS="-j4" ./retropie-setup.sh

joolswills, 2015-02-17:
note: some emulators may not build out of the box box when compiling in parallel. I think I've fixed most of them now, but some may still fail.

closed 2015-02-17 by joolswills

sreichholf, 2015-02-17:
Ah, missed that option, thx for the info!

joolswills, 2015-02-17:
it may be enabled by default in the future once we are 100% sure everything is completely working - and admittedly it isn't really documented yet - no-one likes writing documentation :)

sreichholf, 2015-02-17:
Knowing about the option is fine for me ;). Everyone else has to start reading issues.
And you're right, no one likes that :D