RetroPie/RetroPie-Setup #665: 4snes 4nes not working since 2.3 retropie

ehpower2, 2015-02-17:
I use one of these for my project to fit inside a nes case: http://www.raphnet-tech.com/products/4nes4snes/

Since the update to 2.4 it has stopped seeing all they joystick axis and each controller only will use the first two axis (0 and 1) also the whole control scheme seems more reversed when you try and assign it to one of the input configs in retroarch. I really need this part working to allow my un modded nes and snes controllers to work again as i still use them on the original consoles.

joolswills, 2015-02-18:
none of us have one of these devices for testing though, so it might be better to see if anyone else on the forum / raspberry pi forum has an idea. it's likely related to a kernel update.

Perhaps you can try and see if it works with an older kernel and track down when it stopped working

http://tech.enekochan.com/en/2014/03/08/upgradedowngrade-to-a-specific-firmware-kernel-version-with-rpi-update-in-raspbian/

ehpower2, 2015-02-18:
Ok the big question is if i were to go back to the 2.3 version of retropie  where i last found it worked that leaves two questions.

One: can i use the update script portion of retropie_setup.sh to upgrade back to lastest files along with using the apt-get upgrade to allow it to run on pi2?

two: if i upgrade will the new updates leave the 2.3 kernel alone keeping the whole thing still working assuming that fixes my issue?

joolswills, 2015-02-18:
it will upgrade the kernel if going back to 2.3 and updating, or a new rasbian bootloader/kernel package comes out. I wouldn't recommend updating retropie from a 2.3 image anyway.

Please follow the instructions in the link above to downgrade the kernel.

joolswills, 2015-09-25:
A new kernel has come out since this ticket was created - does work with the latest retropie (+ updating packages once installed)

vgplayer, 2015-10-12:
I have a 2nes2usb adapter (basically same firmware) and am having similar results with RetroPi 3.1 and 3.2.  jstest only reports 2 of the 4 axes.  I also got the same result using jstest on Rasbian Jessie.  No issues when using Debian Jessie on x86 machine.  

I did get it working by adding to the line in /boot/cmdline.txt 
usbhid.quirks=0x289b:0x0002:0x040

This made the controller report as 2 separate inputs.

joolswills, 2015-12-26:
4snes4nes is confirmed as working. if using 1.3 firmware you need to add

usbhid.quirks=0x1781:0x0A9D:0x040 to /boot/cmdline.txt

http://blog.petrockblock.com/forums/topic/autofire-on-the-axis/page/7/#post-112516

closed 2015-12-26 by joolswills