RetroPie/RetroPie-Setup #666: Latest runcommand.sh only runs in mode 4

MonkeyWrench32, 2015-02-17:
I'm using the latest runcommand.sh file. I'm very grateful that I'm able to run Atari 2600 and a couple of other emulators in mode 4 but it appears that it's stuck this way for every emulator. I'm trying to run Sega 32X on modes 0 or 1 and it's just not working. I've even tried using modes such as CEA-1 and mode[1-CEA-16:9]. Any ideas?

joolswills, 2015-02-17:
I may have introduced a bug with an update I did. Please can you add "set +x" to the top of runcommand.sh in /opt/retropie/supplementary/runcommand.sh and then from a console do

```bash
/opt/retropie/supplementary/runcommand/runcommand.sh 1 echo "hello" 2>logfile.txt
```

and paste me the log ?

also the output of this would be useful

```
tvservice -m CEA
tvservice -m DMT
```

thanks.

joolswills, 2015-02-17:
did this work before and the functionality has changed, or did it never work correctly? have you the runcommand update from yesterday ?

joolswills, 2015-02-17:
I meant "set -x" above. sorry

MonkeyWrench32, 2015-02-17:
logfile.txt:

```
+ video_conf=/opt/retropie/configs/all/videomodes.cfg
+ dispmanx_conf=/opt/retropie/configs/all/dispmanx.cfg
+ retronetplay_conf=/opt/retropie/configs/all/retronetplay.cfg
+ declare -A mode
+ mode[1-CEA-4:3]=CEA-1
+ mode[1-DMT-4:3]=DMT-4
+ mode[1-CEA-16:9]=CEA-1
+ mode[4-CEA-4:3]=DMT-16
+ mode[4-DMT-4:3]=DMT-16
+ mode[4-CEA-16:9]=CEA-4
+ reqmode=1
+ [[ -z 1 ]]
+ command=echo
+ [[ -z echo ]]
+ emulator=hello
+ [[ -z hello ]]
+ emusave=hello
+ emusave=hello
++ echo echo
++ cut '-d ' -f1
++ md5sum
+ romsave=r53f31a089339194f333d2e3995dbb05e
+ netplay=0
+ get_mode hello r53f31a089339194f333d2e3995dbb05e
+ local emusave=hello
+ local romsave=r53f31a089339194f333d2e3995dbb05e
++ tvservice -s
+ status='state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
+ [[ state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive =~ (PAL|NTSC) ]]
+ mode_cur=($(echo "$status" | grep -oE "(CEA|DMT) \([0-9]+\)"))
++ grep -oE '(CEA|DMT) \([0-9]+\)'
++ echo 'state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
+ mode_cur_type=CEA
+ mode_cur_id=16
+ mode_cur=CEA-16
++ echo 'state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
++ grep -oE '(16:9|4:3)'
+ mode_cur_aspect=16:9
+ [[ -z 16:9 ]]
+ [[ -f /opt/retropie/configs/all/videomodes.cfg ]]
+ [[ -z '' ]]
+ [[ 1 =~ ^(DMT|CEA)-[0-9]+$ ]]
+ [[ 1 =~ ^(PAL|NTSC)-(4:3|16:10|16:9)$ ]]
+ mode_new=CEA-1
+ clear
+ echo 'Press '\''x'\'' or '\''m'\'' to configure launch options for emulator/port (hello)'
+ read -t 1 -N 1 key
+ [[ '' =~ [xXmM] ]]
+ switched=0
+ [[ -n CEA-1 ]]
+ [[ CEA-1 != \C\E\A\-\1\6 ]]
+ switch_mode CEA-1
+ mode=(${1//-/ })
+ local mode
+ local switched=0
+ [[ CEA == \P\A\L ]]
+ [[ CEA == \N\T\S\C ]]
++ grep -w 'mode 1'
++ tvservice -m CEA
+ hasmode='           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive '
+ [[ -n CEA 1 ]]
+ [[ -n            mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive  ]]
+ tvservice -e 'CEA 1'
+ switched=1
+ [[ 1 -eq 1 ]]
+ reset_framebuffer
+ sleep 1
+ fbset -depth 8
+ fbset -depth 16
+ return 1
+ switched=1
+ config_dispmanx hello
+ local name=hello
+ [[ -f /opt/retropie/configs/all/dispmanx.cfg ]]
+ source /opt/retropie/configs/all/dispmanx.cfg
++ dgen=1
++ uae4all=1
++ vice=1
+ [[ '' == \1 ]]
+ echo performance
+ sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
+ retroarch_append_config
+ [[ ! echo =~ retroarch ]]
+ return
+ eval echo
++ echo
+ sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
+ echo ondemand
+ [[ 1 -eq 1 ]]
+ restore_mode CEA-16
+ mode=(${1//-/ })
+ local mode
+ [[ CEA == \P\A\L ]]
+ [[ CEA == \N\T\S\C ]]
+ tvservice -p
+ reset_framebuffer
+ sleep 1
+ fbset -depth 8
+ fbset -depth 16
+ exit 0
```

Mode switching worked before on some systems, but not all. I couldn't get the 2600 to run mode 4 but I was able to change GB from 0 to 4. My original runcommand.sh was the one that came on the RetroPie 2.4 beta SD card image.

joolswills, 2015-02-17:
the log above looks like it switched to CEA-1 fine for running echo. it ran "tvservice -e 'CEA 1'"

joolswills, 2015-02-17:
can you give me a non working example - when something doesnt switch as expected ? in the same fashion ?

MonkeyWrench32, 2015-02-17:
With the SNES emulator, if I change the mode to 0 or 1, it's still in 4. It says at the top of the screen when I launch a game it's running CEA-4.

joolswills, 2015-02-17:
please change /etc/es_system.cfg and append "2>/home/pi/somelog.txt" to the runcommand parameter for that emulator and provide me with that log again. thanks.

how are you changing the mode btw ? using the gui ?

MonkeyWrench32, 2015-02-17:
SNES with runcommand 0:
```
+ video_conf=/opt/retropie/configs/all/videomodes.cfg
+ dispmanx_conf=/opt/retropie/configs/all/dispmanx.cfg
+ retronetplay_conf=/opt/retropie/configs/all/retronetplay.cfg
+ declare -A mode
+ mode[1-CEA-4:3]=CEA-1
+ mode[1-DMT-4:3]=DMT-4
+ mode[1-CEA-16:9]=CEA-1
+ mode[4-CEA-4:3]=DMT-16
+ mode[4-DMT-4:3]=DMT-16
+ mode[4-CEA-16:9]=CEA-4
+ reqmode=0
+ [[ -z 0 ]]
+ command=echo
+ [[ -z echo ]]
+ emulator=hello
+ [[ -z hello ]]
+ emusave=hello
+ emusave=hello
++ cut '-d ' -f1
++ echo echo
++ md5sum
+ romsave=r53f31a089339194f333d2e3995dbb05e
+ netplay=0
+ get_mode hello r53f31a089339194f333d2e3995dbb05e
+ local emusave=hello
+ local romsave=r53f31a089339194f333d2e3995dbb05e
++ tvservice -s
+ status='state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
+ [[ state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive =~ (PAL|NTSC) ]]
+ mode_cur=($(echo "$status" | grep -oE "(CEA|DMT) \([0-9]+\)"))
++ grep -oE '(CEA|DMT) \([0-9]+\)'
++ echo 'state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
+ mode_cur_type=CEA
+ mode_cur_id=16
+ mode_cur=CEA-16
++ echo 'state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
++ grep -oE '(16:9|4:3)'
+ mode_cur_aspect=16:9
+ [[ -z 16:9 ]]
+ [[ -f /opt/retropie/configs/all/videomodes.cfg ]]
+ [[ -z '' ]]
+ [[ 0 =~ ^(DMT|CEA)-[0-9]+$ ]]
+ [[ 0 =~ ^(PAL|NTSC)-(4:3|16:10|16:9)$ ]]
+ mode_new=
+ clear
+ echo 'Press '\''x'\'' or '\''m'\'' to configure launch options for emulator/port (hello)'
+ read -t 1 -N 1 key
+ [[ '' =~ [xXmM] ]]
+ switched=0
+ [[ -n '' ]]
+ config_dispmanx hello
+ local name=hello
+ [[ -f /opt/retropie/configs/all/dispmanx.cfg ]]
+ source /opt/retropie/configs/all/dispmanx.cfg
++ dgen=1
++ uae4all=1
++ vice=1
+ [[ '' == \1 ]]
+ sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
+ echo performance
+ retroarch_append_config
+ [[ ! echo =~ retroarch ]]
+ return
+ eval echo
++ echo
+ sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
+ echo ondemand
+ [[ 0 -eq 1 ]]
+ reset_framebuffer
+ sleep 1
+ fbset -depth 8
+ fbset -depth 16
+ exit 0
```

With runcommand 1:
```
+ video_conf=/opt/retropie/configs/all/videomodes.cfg
+ dispmanx_conf=/opt/retropie/configs/all/dispmanx.cfg
+ retronetplay_conf=/opt/retropie/configs/all/retronetplay.cfg
+ declare -A mode
+ mode[1-CEA-4:3]=CEA-1
+ mode[1-DMT-4:3]=DMT-4
+ mode[1-CEA-16:9]=CEA-1
+ mode[4-CEA-4:3]=DMT-16
+ mode[4-DMT-4:3]=DMT-16
+ mode[4-CEA-16:9]=CEA-4
+ reqmode=1
+ [[ -z 1 ]]
+ command=echo
+ [[ -z echo ]]
+ emulator=hello
+ [[ -z hello ]]
+ emusave=hello
+ emusave=hello
++ md5sum
++ cut '-d ' -f1
++ echo echo
+ romsave=r53f31a089339194f333d2e3995dbb05e
+ netplay=0
+ get_mode hello r53f31a089339194f333d2e3995dbb05e
+ local emusave=hello
+ local romsave=r53f31a089339194f333d2e3995dbb05e
++ tvservice -s
+ status='state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
+ [[ state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive =~ (PAL|NTSC) ]]
+ mode_cur=($(echo "$status" | grep -oE "(CEA|DMT) \([0-9]+\)"))
++ grep -oE '(CEA|DMT) \([0-9]+\)'
++ echo 'state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
+ mode_cur_type=CEA
+ mode_cur_id=16
+ mode_cur=CEA-16
++ grep -oE '(16:9|4:3)'
++ echo 'state 0x12001a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive'
+ mode_cur_aspect=16:9
+ [[ -z 16:9 ]]
+ [[ -f /opt/retropie/configs/all/videomodes.cfg ]]
+ [[ -z '' ]]
+ [[ 1 =~ ^(DMT|CEA)-[0-9]+$ ]]
+ [[ 1 =~ ^(PAL|NTSC)-(4:3|16:10|16:9)$ ]]
+ mode_new=CEA-1
+ clear
+ echo 'Press '\''x'\'' or '\''m'\'' to configure launch options for emulator/port (hello)'
+ read -t 1 -N 1 key
+ [[ '' =~ [xXmM] ]]
+ switched=0
+ [[ -n CEA-1 ]]
+ [[ CEA-1 != \C\E\A\-\1\6 ]]
+ switch_mode CEA-1
+ mode=(${1//-/ })
+ local mode
+ local switched=0
+ [[ CEA == \P\A\L ]]
+ [[ CEA == \N\T\S\C ]]
++ tvservice -m CEA
++ grep -w 'mode 1'
+ hasmode='           mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive '
+ [[ -n CEA 1 ]]
+ [[ -n            mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive  ]]
+ tvservice -e 'CEA 1'
+ switched=1
+ [[ 1 -eq 1 ]]
+ reset_framebuffer
+ sleep 1
+ fbset -depth 8
+ fbset -depth 16
+ return 1
+ switched=1
+ config_dispmanx hello
+ local name=hello
+ [[ -f /opt/retropie/configs/all/dispmanx.cfg ]]
+ source /opt/retropie/configs/all/dispmanx.cfg
++ dgen=1
++ uae4all=1
++ vice=1
+ [[ '' == \1 ]]
+ sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
+ echo performance
+ retroarch_append_config
+ [[ ! echo =~ retroarch ]]
+ return
+ eval echo
++ echo
+ sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
+ echo ondemand
+ [[ 1 -eq 1 ]]
+ restore_mode CEA-16
+ mode=(${1//-/ })
+ local mode
+ [[ CEA == \P\A\L ]]
+ [[ CEA == \N\T\S\C ]]
+ tvservice -p
+ reset_framebuffer
+ sleep 1
+ fbset -depth 8
+ fbset -depth 16
+ exit 0
```

I've been editing /etc/emulationstation/es_systems.cfg directly. I wasn't even aware of a GUI option.

joolswills, 2015-02-17:
the above looks fine.

runcommand 0 means do not change the resolution. that's what it does in the log above.
runcommand 1 chooses CEA-1 and switches.

Can't find any reference to switching to CEA-4

The logs above are from running "echo" manually though right? I need the output of runcommand when launching from emulation station when you say it says about mode-4. should be able to redirect the output with "2>/home/pi/somelog.txt" by adding that to the launch command in es_systems.cfg

you can change videomode by pressing "m" when launching an emulator. it says at the top of the screen before launching.

joolswills, 2015-02-17:
or if you added "hello" to es_system.cfg that's not what I wanted, just the 2>/home/pi/somelog.txt on the end to get the output. the rest should be left the same

MonkeyWrench32, 2015-02-17:
I just did some testing on my own. Apparently the changes for es_systems.cfg only take effect when my Pi is rebooted. After making changes and then rebooting, things are working as expected. If this is standard behavior, I am terribly sorry for wasting your time. :(

joolswills, 2015-02-17:
it should work if you exit and restart emulation station also.

you shouldn't need to edit the config for changing screenmodes. there are defaults in there, but you can override them per emulator, or even per rom, by pressing "m" on emu startup. 

Don't worry - I'm actually relieved there isn't a bug since I built some new images last night with this code.

closed 2015-02-17 by joolswills

MonkeyWrench32, 2015-02-17:
Thanks for all of your help. At least I was able to be useful as a tester. :)