RetroPie/RetroPie-Setup #669: .vice dir missing in 2.5.0 beta

winstonoboogie, 2015-02-18:
Had to manually create the directory /home/pi/.vice 

If the directory doesn't exist, Vice cannot save joystick-mapping...

joolswills, 2015-02-18:
sounds like a vice bug, since we use a different location for our vice config - I guess vice shoud handle making that folder itself if it needs it for other configs. I'll either create the folder, or see if we can get the joystick configs saving to the same location as the vice config.

winstonoboogie, 2015-02-18:
OK. Since Vice got its own directory under /opt/retropie/emulators since version 2.4 I assumed it had something to do with that... In version 2.4 /home/pi/.vice exists from scratch...

joolswills, 2015-02-18:
vice usually creates the folder with its config in on the first run. it's possible I left the config folder there by accident or we hadn't moved the config at v2.4. I hadn't realised vice had additional config file for mapping, and would want to use the ~/.vice folder. It should really create it as it does for its main config though in that case.

thanks for the heads up.

joolswills, 2015-02-18:
fixed.

closed 2015-02-18 by joolswills