RetroPie/RetroPie-Setup #670: add Adventure Games Studio (AGS) support label:enhancement

jagiella, 2015-02-18:
Recently I discovered that the Adventure Games Studio (http://www.adventuregamestudio.co.uk/) also provided a port for Debian / Raspbian (https://github.com/adventuregamestudio/ags/blob/master/debian/README.md). Its a SCUMM-like engine running a wide range of indie classic-style adventures as for example the ones commercially published by wadjet eye (www.wadjeteyegames.com). It would be great to have that engine become a part of the retropie familie. 

Did I convince you considering? :)

free5ty1e, 2015-02-19:
Awesome, at the very least I'm going to be installing this on my RetroPie.  Great find, thanks!!

joolswills, 2015-04-03:
added as an experimental module.

For AGS games that are 320x240 resolution, you will want to use the new "runcommand.sh" option to switch framebuffer resolution, which will allow you to run them fullscreen.

HerbFargus, 2015-04-04:
Has anyone gotten sound to work in their games? 

I've tested a few and the games work but there is no sound, so I'm not sure if it is my games that are broken (i don't think they are) of if it is just something I'm missing.

The readme says something about a patch:

https://github.com/adventuregamestudio/ags/blob/master/debian/README.md

and this thread says something about needing pulseaudio 

http://www.adventuregamestudio.co.uk/forums/index.php?topic=46149.0

jagiella, 2015-04-05:
Great news!! I pulled and installed the changes including the experimental AGS package. I tested "Over The Edge" which starts nicely, but with a tiny image due to its small image resolution.

@joolswills could you explain how to run it in fullscreen? Where and which changes do I need to do in the config in order to start the AGS-engine in fullscreen despite low native resolution? Some tutorial would be great. :)

joolswills, 2015-04-05:
@HerbFargus I tried a couple of games and both had sound, so not sure. can you give me an example which doesnt ?

jagiella, 2015-04-05:
@joolswills pushing 'x'/'m' key and switching the framebuffer resolution to the maximally possible 1280x960 still leads to a small image size.

@HerbFargus I tested several games myself and didn't encounter any sound problems. So maybe your system has some sound issues.

joolswills, 2015-04-05:
you need to lower the resolution. so if the game is 320x240, use 320x240 for the framebuffer res.

jagiella, 2015-04-05:
@joolswills well that worked indeed :) Thank you so much! Now the game runs perfectly in fullscreen and even without any notable slow down. Looking forward to see this engine in upcoming RetroPie images...with a nice skin for emulationstation :)

HerbFargus, 2015-04-06:
Sorry for the delay, I was testing a bunch. I tried some demos from wadjeteye like resonance http://www.wadjeteyegames.com/games/resonance/ and it threw an error saying that  
```
ERROR: The file is corrupt. Make sure you have the correct version of the editor, and that this really is an AGS game.
```
I also get the same issue from a bunch of games at this site: http://www.adventuregamestudio.co.uk/site/games/game/644/

This game runs but has no sound 
http://www.adventuregamestudio.co.uk/site/games/game/1851/
and many other similar to it.

So I'm not sure if its just my source of games, or what. I tested with my Rpi2 with a fresh 3.0 BETA 2
image.

joolswills, 2015-06-20:
It's possible not all games work, but at least some i tested did. Closing for now.

closed 2015-06-20 by joolswills