RetroPie/RetroPie-Setup #673: Keyboard doesn't work in game with a joystick plugged in

igorissen, 2015-02-20:
Hi,

I have just get my Raspberry Pi2 and install RetroPie. I have also added Street Fighter 2 Snes rom to play with a friend but when we start to play the keyboard doesn't respond at all.

I have this configuration in my `retroarch.cfg`

```
...
...
input_player1_a = s
input_player1_b = d
input_player1_y = z
input_player1_x = e
input_player1_start = enter
input_player1_select = rshift
input_player1_l = a
input_player1_r = r
input_player1_left = left
input_player1_right = right
input_player1_up = up
input_player1_down = down
...
...
input_player2_joypad = "0"
input_player2_b_btn = "0"
input_player2_y_btn = "2"
input_player2_select_btn = "6"
input_player2_start_btn = "7"
input_player2_up_btn = "h0up"
input_player2_down_btn = "h0down"
input_player2_left_btn = "h0left"
input_player2_right_btn = "h0right"
input_player2_a_btn = "1"
input_player2_x_btn = "3"
input_player2_l_btn = "4"
input_player2_r_btn = "5"
input_player2_l2_axis = "+2"
input_player2_r2_axis = "+5"
input_player2_l3_btn = "9"
input_player2_r3_btn = "10"
input_player2_l_x_plus_axis = "+0"
input_player2_l_x_minus_axis = "-0"
input_player2_l_y_plus_axis = "+1"
input_player2_l_y_minus_axis = "-1"
input_player2_r_x_plus_axis = "+3"
input_player2_r_x_minus_axis = "-3"
input_player2_r_y_plus_axis = "+4"
input_player2_r_y_minus_axis = "-4"
```

When the game start, the keyboard doesn't respond anymore and the joystick (Logitech Chillstream Xbox 360 USB controller) works well.

Any idea where I can put some configuration to enable the keyboard as player 1 ?

Thank you and great work with Retropie.

igorissen, 2015-02-23:
I think I found the issue and I don't know if it is a Retropie or Retroarch issue. When I wrote this issue I have installed the 3 SNES emulators in the "source installation" list, and now I have reinstalled but with only 1 SNES emulator (libretrocore) and now all work very well. All libretrocore emulators use retroarch but other not.

So for those who have the same issue as me you need to installed only libretrocore emulators.

joolswills, 2015-02-23:
oh ok. so the problem was with a non retroarch emulator ? those will need to be configured separately unfortunately - depending on which emulator. thanks for the feedback.

igorissen, 2015-02-23:
@joolswills Yes I think that the keyboard and controller issue is something about the non retroarch emulators. I also had this issue when I give a try with PiPlay. I have switched to Retropie since I did not found a solution for that problem.

Thanks to the retroarch project I can now play with my friends !!

closed 2015-02-23 by igorissen