RetroPie/RetroPie-Setup #674: Manual patch description to remove mouse cursor in AdvMame:

andrewsil1, 2015-02-21:
Since the AdvMame source isn't actually part of this repo, I'll just provide the appropriate chunk of code... Verified working with current 0.94 source from the setup script in tmp/build.

In advmame/advance/sdl/vsdl.c:
```c
 switch (sdl_state.cursor) {
        case adv_cursor_auto :
                /* if ((sdl_state.surface->flags & SDL_FULLSCREEN) != 0) { // Added block comment here
                        SDL_ShowCursor(SDL_DISABLE);
                } else {
                        SDL_ShowCursor(SDL_ENABLE);
                } */  //Ended here
                SDL_ShowCursor(SDL_DISABLE); // Added this line.
                break;
        case adv_cursor_on :
/*              SDL_ShowCursor(SDL_ENABLE); //Added block comment here.
                break; */ //Ended here so that both on/off cases disable the cursor.
        case adv_cursor_off :
                SDL_ShowCursor(SDL_DISABLE);
                break;

joolswills, 2015-02-22:
in ~/.advance/advmame.rc what happens if you set "device_video_cursor" to off. That turns off the cursor for me without any code changes.

andrewsil1, 2015-02-22:
Ugh, you're totally right!  Don't know why I didn't see that sooner.  Thanks for the suggestion, no patch needed.

closed 2015-02-22 by andrewsil1

joolswills, 2015-02-22:
I will change the default config to include this. Cheers.

andrewsil1, 2015-02-22:
Cheers to you- thanks for all the hard work.

_____________
Andy