RetroPie/RetroPie-Setup #677: [Feature Request] Add PPSSPP Emulator

robertybob, 2015-02-22:
This has been successful built on a RPi2 
[Link](https://github.com/hrydgard/ppsspp/issues/7479#issuecomment-75403013)

>Got it working on RPI2.

>You don't need to change much:

>you need to get the EGL, GLES, GLESv2 and KHR headers from the mesa packages, preferably put them in the ppsspp source folder
>change CMakeLists.txt and remove RPI references: ppssppsdl uses SDL2 which already has videocore backend, so we don't need any rpi-specific backend here
>add the egl/gles include folder (where you copied the EGL, GLES, GLESv2 and KHR folders from mesa) and the egl/gles lib folder (the rpi one - /opt/vc/lib) - you can see in CMakeLists.txt how it was done previously for rpi
>Run cmake, edit CMakeCache.txt and set FBDEV to ON, GLES2 to ON, maybe even MOBILE_DEVICE to ON, run cmake again, then run make.

>A bit disappointed with the performance (GPU clocked at 500MHz), it runs fullspeed on the odroid... 
>https://www.youtube.com/watch?v=O8juShfoJtE

If this runs better than the current N64 emulators then surely it's worth including?

robertybob, 2015-02-23:
@joolswills ? :)

joolswills, 2015-02-25:
We don't always have time to respond to all tickets immediately. I had read this, but haven't had time to look into it further yet.

julianxhokaxhiu, 2015-02-27:
This would be easy for you to include I think https://github.com/libretro/libretro-ppsspp

GAMELASTER, 2015-03-01:
libretro dont work :) tested

barbudreadmon, 2015-03-05:
Actually, libretro-ppsspp works if you tweak the sources a little, but is too slow.

GAMELASTER, 2015-03-07:
Barbudreadmon: i looked some videos and its possible, and its not so much slow :-) but i didnt get idea how to do it.. mmm

barbudreadmon, 2015-03-07:
You are talking about ppsspp-libretro or ppssppsdl ? I know ppssppsdl run at (near ?) full speed, but i'm not sure the same is possible on libretro-ppsspp at the moment.

joolswills, 2015-10-26:
both ppsspp and the libretro port are now available from the experimental menu (update retropie-setup script first). Cheers!

closed 2015-10-26 by joolswills

julianxhokaxhiu, 2015-11-07:
Awesome! Thank you